MagicMirrorOrg/MagicMirror #570: Mirror Freezing

nhubbard, 2016-12-16:
When submitting a new issue, please supply the following information:

**Platform**: 
* **Electron** Latest
* **Node** Latest
* **Raspberry Pi 3**

**MagicMirror Version**: Version 2

**Description**: Over a small amount of time after starting, the mirror will work fine until seemingly random intervals of time occur, in which the mirror freezes, along with the host OS. This also disallows SSH until the system is hard reset.

**Steps to Reproduce**: Using a stock Raspberry Pi 3, set up the system with the normal setup and the snow module. The system should freeze. **No spike in CPU or RAM usage was observed!**

**Expected Results**: The system works as normal.

**Actual Results**: The system freezes, causing the need for a hard reset.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information! It looks like the basic config with the snow module.

nhubbard, 2016-12-16:
I have checked the snow module, and it does still freeze without it.

roramirez, 2016-12-16:
Hi @nhubbard 

What is snow module?

Jopyth, 2016-12-16:
@roramirez Can be found here: https://github.com/nhubbard/MagicPlugins/tree/master/snow

MichMich, 2016-12-19:
Make sure you use a powerfull and stable powersupply. Try an other one to see if that solves the issue.

roramirez, 2016-12-21:
Thanks @Jopyth

I 'm testing  the snow module. It's very funny for the time ... well I'm in the other side of world. Here dont snow at this time of year.

@nhubbard I sent a PL for getting the  images from local side. Maybe could be this.

nhubbard, 2016-12-23:
Yeah, I fixed it. Turns out forgetting to install the new dependencies won't break it, but will make it freeze constantly. :-1:

nhubbard, 2016-12-23:
It hasn't frozen for about 8 days now.

closed 2016-12-23 by nhubbard