MagicMirrorOrg/MagicMirror #571: Can't access module public folder

mrvanes, 2016-12-17:
I've created a copy of HelloWorld module called pvoutput and created a sub directory public in which I created a file test.txt. I assumed this file would be available on localhost:8080/pvoutput/public/test.txt but all I get is a "Cannot GET /pvoutput/public/test.txt".

Jopyth, 2016-12-18:
Try `localhost:8080/modules/pvoutput/test.txt` or `localhost:8080/modules/pvoutput/public/test.txt`.

Also for development support, a thread in [the forum section for development](https://forum.magicmirror.builders/category/6/development) instead of an issue on github might be better.

mrvanes, 2016-12-19:
That worked, and thx for the link to the forum! Will pick up furthere inquiries there!

closed 2016-12-19 by mrvanes