MagicMirrorOrg/MagicMirror #573: Module CurrentWeather

fry0815, 2016-12-18:
The module is not loading, if the given temperature from openweather.org is zero (0).
The problem is the check in line 221: 
```javascript
if (!data || !data.main || !data.main.temp) {
	// Did not receive usable new data.
	// Maybe this needs a better check?
	return;
}
```
If `data.main.temp` is 0, it goes to `return`. Yeah you are right, it needs a better check. :)

For example i fixed the problem like this:
```javascript
if (!data || !data.main || typeof data.main.temp === 'undefined') {
	// Did not receive usable new data.
	// Maybe this needs a better check?
	return;
}
```

I hope this will help you.

roramirez, 2016-12-18:
@fry0815  Why dont send a Pull Request fixed this situation?.

The new contributions are welcome :)

fry0815, 2016-12-19:
@roramirez I dont know, I'm new in this situation on a public project. I try to do this at home.
Thank you :)

nhubbard, 2017-01-06:
@fry0815 I can make a pull request if you wish to allow me to.

fry0815, 2017-01-10:
Oh sry, atm i have no time. @nhubbard yes can do it! Thank you! :)

closed 2017-01-10 by MichMich

roramirez, 2017-01-10:
Hi @MichMich, I think this issue need stay open ;)

Paulismo, 2017-07-31:
Hello everybody. I have an issue in the currentweather module, my time is freezing, it starts from 20:xx and stops right there, i haven`t iven a idea about what`s causing this issue and i hope you`ll help me out. I`m working with raspberry pi btw.

fewieden, 2017-07-31:
isn't the time in current weather just a disolay for sunset and sunrise?

Paulismo, 2017-08-01:
Oh, thank you for support, I just couldn't figure this problem out beckause
lack of information, but anyway you helped me a lot.