MagicMirrorOrg/MagicMirror #575: Electron fails to start due to missing libgtk dependency

mattcarrier, 2016-12-19:
Raspberry PI 3

1. Installed Raspbian Jessie Lite
2. Ran magicmirror curl install
3. Ran start command as listed below

Raspbian Jessie Version
```
Version:November 2016
Release date:2016-11-25
Kernel version:4.4
```

latest sha: b8167f89e54a71a697cc1e25e2b524ce95a410a5

```
pi@magicmirror:~/MagicMirror $ DISPLAY=:0 npm start

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/electron: error while loading shared libraries: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory

npm ERR! Linux 4.4.34-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.9.2
npm ERR! npm  v3.10.9
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
```

mattcarrier, 2016-12-19:
I installed raspberrypi-ui-mods, libxss1, libnss3 and now have the following problem.  Is NOOBS or Raspbian with Pixel required, did I miss that somewhere in the docs?

```
pi@magicmirror:~/MagicMirror $ DISPLAY=:0 npm start

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

npm ERR! Linux 4.4.34-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.9.2
npm ERR! npm  v3.10.9
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
pi@magicmirror:~/MagicMirror $ cat npm-debug.log 
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.9
3 info using node@v6.9.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
6 silly lifecycle magicmirror@2.0.0~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0
8 verbose lifecycle magicmirror@2.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.0.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.0.0~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.0.0~start: Args: [ '-c', 'electron js/electron.js' ]
12 silly lifecycle magicmirror@2.0.0~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.0.0~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.0.0 start: `electron js/electron.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:877:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid magicmirror@2.0.0
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.4.34-v7+
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v6.9.2
20 error npm  v3.10.9
21 error code ELIFECYCLE
22 error magicmirror@2.0.0 start: `electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     electron js/electron.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

MichMich, 2016-12-19:
Jessie Lite is not officially supported. For support on this matter, please visit the forum: http:/forum.magicmirror.builders

mattcarrier, 2016-12-19:
@MichMich thanks for getting back to me.  That's fine, I was unable to find anything that explicitly states what is required as far as OS dependency.  Anything you would be able to point me to?

MichMich, 2016-12-20:
Check out the regular Raspbian OS (so not the light version)

paviro, 2016-12-20:
Or try: 
sudo apt-get install raspberrypi-ui-mods

paviro, 2016-12-20:
Looks like you also need: 
sudo apt-get install xinit
Probably easier to do as Michael said 😊

nhubbard, 2017-01-06:
We now have an official guide to installing on Jessie Lite [here](https://github.com/MichMich/MagicMirror/wiki/Jessie-Lite-Installation-Guide). I highly recommend that you use that if you want to use Jessie Lite.

closed 2017-01-10 by MichMich