MagicMirrorOrg/MagicMirror #576: deleted weekly reoccurring calendar items still showing in calendar

dinkybluebug, 2016-12-20:
**Platform**: RPI 3

**Node Version**: 6

**MagicMirror Version**: 2

**Description**: Weekly recurring calendar items that have been deleted, are still showing in the calendar - which is odd as when I view my icloud calendar in my phone, my partners phone, online, and in outlook the items are not there.  But they are definitely showing still in my magicmirror calendar.

Ive restarted - but this doesnt fix issue.  Ive added in new items into the calendar, and these show - so its definitely working - not sure why deleted items  are still showing.

Any ideas as to why this might be?

roramirez, 2016-12-20:
When you download the calendar by url the events are still there?

FatBasta, 2016-12-20:
I have the same issue.

Platform: RPI 3
Node Version: 6
MagicMirror Version: 2
Ical from google calendar

It seems that if you have a recurring event and remove one from the middle, it still shows in the mirror.
For example a recurring event for weeks 18 - 25, and you delete week 23, google calendar no longer shows it, but the mirror still does.

dinkybluebug, 2016-12-20:
Yes that's what's happened to me. One event in the middle of a reoccurring appointment was deleted but it still shows on mirror - even though Apple calendar doesn't show it

roramirez, 2016-12-20:
Hi!, some way to debug the problem is what happend with data into url you called. Maybe the events still there and flagged "deleted" or something like data.

I think a good way is downloaded the file .ical the moment this happend.

morozgrafix, 2016-12-31:
I believe skipped recurring events (ones that have EXDATE timestamp) aren't skipped in ical.js https://github.com/peterbraden/ical.js/issues/75 which leads to problem in rrule dependency (which may be unmaintaned).

Here is a part of my ics file that has 2 skipped events from the recurring series. They still show up in MM (note the EXDATE values for 12/24 and 12/31 which were "deleted" events):
```
BEGIN:VEVENT
DTSTART;TZID=America/Los_Angeles:20160924T083000
DTEND;TZID=America/Los_Angeles:20160924T093000
EXDATE;TZID=America/Los_Angeles:20161224T083000
EXDATE;TZID=America/Los_Angeles:20161231T083000
RRULE:FREQ=WEEKLY
DTSTAMP:20161231T105941Z
UID:FCE045E3-5271-442E-9E96-D5A1DE0958D3
CREATED:20160917T153742Z
DESCRIPTION:
LAST-MODIFIED:20161231T022115Z
LOCATION:Cahill's Judo Academy\n635 San Mateo Ave\nSan Bruno CA 94066\nUnit
 ed States
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Judo
TRANSP:OPAQUE
X-APPLE-STRUCTURED-LOCATION;VALUE=URI;X-ADDRESS=635 San Mateo Ave\\nSan Bru
 no CA 94066\\nUnited States;X-APPLE-MAPKIT-HANDLE=CAESpwIIrk0Q9+Xw/rvT9ag8G
 hIJPUhPkUPQQkARbm5MT1iaXsAitAEKDVVuaXRlZCBTdGF0ZXMSAlVTGgpDYWxpZm9ybmlhIgJD
 QSoJU2FuIE1hdGVvMglTYW4gQnJ1bm86BTk0MDY2QhJEb3dudG93biBTYW4gQnJ1bm9SDVNhbiB
 NYXRlbyBBdmVaAzYzNWIRNjM1IFNhbiBNYXRlbyBBdmVqBDM2MTGKAQ5TYW4gQnJ1bm8gUGFya4
 oBEkRvd250b3duIFNhbiBCcnVub6IBCjk0MDY2LTM2MTEqFUNhaGlsbCdzIEp1ZG8gQWNhZGVte
 TIRNjM1IFNhbiBNYXRlbyBBdmUyFFNhbiBCcnVubywgQ0EgIDk0MDY2Mg1Vbml0ZWQgU3RhdGVz
 ;X-APPLE-RADIUS=25.6186685884713;X-APPLE-REFERENCEFRAME=1;X-TITLE=Cahill's 
 Judo Academy:geo:37.627062,-122.411640
X-APPLE-TRAVEL-ADVISORY-BEHAVIOR:AUTOMATIC
BEGIN:VALARM
ACTION:NONE
TRIGGER;VALUE=DATE-TIME:19760401T005545Z
X-WR-ALARMUID:150C8954-AE40-42C9-A192-A55BB3F2FE72
UID:150C8954-AE40-42C9-A192-A55BB3F2FE72
ACKNOWLEDGED:20161126T164657Z
END:VALARM
END:VEVENT
```

Jopyth, 2017-01-05:
Related issue and another example: #565

MichMich, 2017-01-10:
Somebody willing to send a PR?

MichMich, 2017-01-12:
Please continue the discussion in #565

closed 2017-01-12 by MichMich