MagicMirrorOrg/MagicMirror #577: electron.js issue

raspberrysauce, 2016-12-22:
I keep getting the following error when trying to run my MM. I've updated everything multiple times.

pi@HamsMagicMirror:~/MagicMirror $ npm start

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v7.3.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

nhubbard, 2016-12-23:
@raspberrysauce Which RPi are you running? One, two, or three?

raspberrysauce, 2016-12-23:
Pi 3!

jayrod, 2016-12-23:
I am getting the same error messages using Pi 2 Model B v1.1 
node v6.9.2
npm v3.10.9

jayrod, 2016-12-24:
I also have the same problem on my Pi 3 Model B with the same build process.

roramirez, 2016-12-24:
Maybe related with #575

frdteknikelektro, 2016-12-24:
I'm just trying MagicMirror2 few days ago, and have same problem.
Okay, here is my temporary solution on commit b8167f8 (Sep 20) master branch. Hope this will fix on new commit.

If you install it from curl, and failed,
```bash
cd ~/MagicMirror
nano package.json
```
Change grunt dependency
```json
"devDependencies": {
    "grunt":"~0.4.0",
```
reinstall dependency, and then built electron
```bash
cd ~/MagicMirror
npm install
cd ~/MagicMirror/node_modules/electron-prebuilt/
node install.js
```
this will take a long time, and finally
```bash
cd ~/MagicMirror
npm start
```

jayrod, 2016-12-24:
My issue is fixed. Raspian lite doesn't come with x11 installed :/  so no gui means no worky.

mzawlocki, 2017-01-08:
Thank you @frdteknikelektro. This fixed my issue.

raspberrysauce, 2017-01-08:
I realized it doesn't work through SSH. When I ran it directly from the pi
terminal it worked just fine
Very strange ! For now , I'm going to continue running it directly .

frdteknikelektro, 2017-01-08:
Seems like new release [v2.1.0](https://github.com/MichMich/MagicMirror/releases/tag/v2.1.0) has fixed this issue, better to close this issue. Happy new year! (?)

it says

> - Replaced `electron-prebuilt` package with `electron` in order to fix issues that would happen after 2017.

RBM-PITT, 2017-01-08:
Does anything have to change in the install to include the new release - I did an install yesterday using the jessielite install guide, it installedl the new 2.1.0; yet I'm getting this ELIFECYCLE error on electron.js

roramirez, 2017-01-08:
@RBM-PITT Are you try this guide [Jessie-Lite-Installation-Guide](https://github.com/MichMich/MagicMirror/wiki/Jessie-Lite-Installation-Guide)?

roramirez, 2017-01-08:
... and ... please check this comment https://github.com/MichMich/MagicMirror/issues/145#issuecomment-271075358

RBM-PITT, 2017-01-08:
Roramirez - thank you for the reply. Yes - I used the Jessie-Lite-Installation-Guide.

I read the comment #145 and decided to try the full install from this tutorial:
https://forum.magicmirror.builders/topic/236/complete-setup-tutorial

I was successful getting it to run with the full install. I look forward to learning more about this application and building my magic mirror!

roramirez, 2017-01-08:
@raspberrysauce through SSH you need run 
`DISPLAY=:0 nohup npm start`

So, this issue it seems need to close.

simplearetenir, 2017-01-24:
Hello,

I have the same issue :
`npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node vXXX
npm ERR! npm vXXX
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.0 start: electron js/electron.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR! electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR! npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR! /home/pi/MagicMirror/npm-debug.log`

I followed the automatic installation procedure, on a raspberry 3 with the last Raspbian Jessie.
I test the command : `sudo npm install` but I always get the error.
I tried several installs (raspian & node) but still the same error.

Can anyone help me?

MichMich, 2017-01-24:
@simplearetenir What version of Jessie are you running?

Maybe we should include a test in the Installer to check of it's the right OS (no Jessie Lite), to make sure everything is ok ... @roramirez Are you interested in setting this up?

gjb2048, 2017-01-24:
Hi I'm not sure its related...

I've been trying Magic Mirror on Xubuntu 16.10 with the 'serveronly' option and found that the debian package was out of date in terms of npm version.  As the raspberry Pi is also debian based (I think) then could try installing 7.4 another way.  I found that uninstalling npm then using the Node.js v7 way here: https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions - worked for me to get MM up and running.  Thus, just the lines:

````
curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash -
sudo apt-get install -y nodejs
````

I only have a Pi B+ so not tried on that yet.

simplearetenir, 2017-01-24:
@MichMich : I run the las version of Jessie (no Jessie lite)

roramirez, 2017-01-24:
Yes @MichMich I can do it.

roramirez, 2017-01-25:
To remember me... please reopen it and assign to me.

MichMich, 2017-01-25:
@roramirez Somehow GitHub wont let me assign it to you.

Jopyth, 2017-01-25:
@MichMich I think you can only assign to collaborators or whatever they are called. Or @roramirez can do it by himself I think. [Source](https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/)

roramirez, 2017-01-25:
Yes, only can set to collaborators or by default user who is open the issue. It's enough with open state with this I can remember ;)

alostvagabond, 2017-01-25:
Ran into the same problem earlier, my solution to it was to just add "sudo" before the actual installer

`sudo curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash`

simplearetenir, 2017-01-26:
I can not do it :
 - I mount a SD card with the latest version of Jessie (11/01/2017).
 - I put it in a raspberry 3 (I also tried in a raspberry 2)
 - I connect ssh to the raspberry
 - I launch the command : "sudo curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash"
Then I launch directly on the raspberry, the CLI command line (not in PIXEL) :
 - I launch "cd Magicmirror"
 - I launch "npm start"

And I always get the same error :
`npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.9.4
npm ERR! npm v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.0 start: electron js/electron.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR! electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR! npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR! /home/pi/MagicMirror/npm-debug.log`

Can anyone do the test with the latest version of Jessie?

simplearetenir, 2017-01-26:
Maybe someone can share with me a backup of an SD card with MM running on a PI3

roramirez, 2017-01-26:
What Raspberry ISO did you download?

simplearetenir, 2017-01-26:
@roramirez RASPBIAN JESSIE WITH PIXEL : https://www.raspberrypi.org/downloads/raspbian/

roramirez, 2017-01-26:
I just already finished a test on Jessie Little. About of HowTo [Jessie Lite Installation Guide ](https://github.com/MichMich/MagicMirror/wiki/Jessie-Lite-Installation-Guide)  I think is missing some packages to install, I added these :
 xinit xserver-xorg

The log trace about the process, but curious can find in Pastebin http://pastebin.com/YBDW2CB7

roramirez, 2017-01-26:
... so, I'll hope soon finish the test for RASPBIAN JESSIE WITH PIXEL

roramirez, 2017-01-26:
Ok, I tested the Raspbian with Pixel and MagicMirror work by default with the script of installation

**Raspbian with Pixel**
**Version**: January 2017
**Release date**: 2017-01-11

Can find the log of the all process here:  http://pastebin.com/8ftthQqf

Two thing, if used Jessie Little feel free to follow https://github.com/MichMich/MagicMirror/wiki/Jessie-Lite-Installation-Guide 

If MagicMirror not work for you can be two things. 

1.-) X windows system of Debian is not working or not installed. This is required if dont use the serveronly mode.
2.) I suppose by comment expresed here the users  not append `DISPLAY=:0` for sentence start in a  SSH remote session

If you can execute the MagicMirror on terminal by SSH session, must be run **LIKE THIS**
`DISPLAY=:0 npm start`

The Pull Request #652 fixed and  prevent this problem. For future instalations  if not set the _DISPLAY_ the script will do it for you.

I've testing Raspbian with Pixel and Little, Release date: 2017-01-11 and all worked.

@MichMich Please close the issue after close the Pull request.

closed 2017-01-27 by MichMich

simplearetenir, 2017-01-27:
@roramirez : I followed the tutorial for Jessie Lite and it works correctly, thank you very much. Just one thing: **the screen saver does not disable** (perhaps due to the addition of xinit xserver-xorg).

I think I know why, I could not make it work with Jessie with Pixel : I started the "npm start" from the raspberry command line and not from a command line from PIXEL.

roramirez, 2017-01-27:
@simplearetenir  if you remove xscreensaver? or remove by desktop setting into Pixel

celianmissud, 2017-02-03:
Same problem like @simplearetenir. I follow the Jessie Lite Installation guide and it works correctly  but the screensaver doesn't disable with : 

@xset s noblank 
@xset s off 
@xset -dpms

into etc/xdg/lxsession/LXDE/autostart and :

xserver-command=X -s 0 -dpms

into sudo nano /etc/lightdm/lightdm.conf

Any idea ?

celianmissud, 2017-02-11:
Problem solved.
It works fine when I add the 3 parameters into /home/pi/.config/lxsession/LXDE/autostart
Instead of /etc/xdg/lxsession/LXDE/autostart

roramirez, 2017-02-11:
@celianmissud Please add note in wiki to help new users.