MagicMirrorOrg/MagicMirror #578: Wheater Forecast can not be proxied

Dom1n1c, 2016-12-24:
All Addons, that i use, can be proxied using nginx, to use the mm on another device, or to access it from another site.
But: the Weather Elements won't load behind nginx.
Maybe there are some paths that need to be adjusted within the mm installation?

Thanks.

MichMich, 2016-12-25:
There currently is no proxy support. Feel free to send a PR.

closed 2016-12-25 by MichMich

Dom1n1c, 2016-12-25:
The ApiBase needs to be proxied in order to allow https access.
This could be done via nginx or similar.
Then you just need to change the apibase in the two configs for current and the weather forecast,

apiBase: "https://nginx.de/openweathermap/", 
which will serve the API via https