MagicMirrorOrg/MagicMirror #587: ipWhitelist - cant be whitelisted

bjornarfonn, 2017-01-02:
After updating to v2.1.0, i cant access this from remote location anymore. Normally accessing devices via VPN to home network, but gives me the default error message on the MM pi.
Also tried adding the IP from remote location without success. 
Simply adding "::" or "0.0.0.0" in config.js doesnt give me access to it.. VNC works, so its running fine on localhost

Is there a way to disable this feature, or am i adding the ip wrong? (Its on double quotes im assuming?)

MichMich, 2017-01-02:
Check your node console output, it should show the blocked ip.

bjornarfonn, 2017-01-02:
Access denied to IP address: ::ffff:10.8.0.6
(VPN adds the extra number "8" on my local ip pool) 
But even trying to add "::ffff:10.8.0.6" or "10.8.0.6" its still displaying the same msg,

roramirez, 2017-01-02:
You need add `::ffff:10.8.0.6` . Will not work if only put the Ip address in IPV4 format.

bjornarfonn, 2017-01-02:
Tried this. And tried on my normal local network using ::ffff:10.0.0.(my device) and ::ffff:10.0.0.2/100 to make all local devices whitelisted, still nothing. Could this be a router issue? Otherwise, i dont know what else to try..

xdker, 2017-01-04:
`ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1","192.168.99.103","192.168.99.202"],`

`Access denied to IP address: ::ffff:192.168.99.202
Access denied to IP address: ::ffff:192.168.99.202
Access denied to IP address: ::ffff:192.168.99.103
Access denied to IP address: ::ffff:192.168.99.103
`

why?

bjornarfonn, 2017-01-04:
@xdker; was able to get access following some advice from the forum.
ipWhitelist: ["::fff:0.0.0.0/1", "::fff:0.8.0.0/1", "::fff:128.0.0.0/2", "::fff:192.0.0.0/3", "::fff:224.0.0.0/4", "127.0.0.1", "::ffff:127.0.0.1", "::1"] 

This bacially allows all variations of IP pools to access the mirror. Im comfortable with having MM accessible on my network. So if you feel the same way, please copy and add to your config.js

closed 2017-01-04 by bjornarfonn