MagicMirrorOrg/MagicMirror #593: MM Freezes up after .......

DRAZINED, 2017-01-03:
I stumble with these MMirror projects and got hook..........I had to have one. Period, so my many thanks to the community.

Project config/set up was fairly easy ....just a matter of careful reading.

My Variables:
RB Pi 3
Install: 2016-11-25-raspbian-jessie
Modules: Defaults, Email Notification, MM Globe, 

My issue: after rebooting MM goes frozen (after 8-11 min)
No scrolling  (RSS) and or Calendar Updates (I tested it by editing calendar events)

No clues where to look to solve the issue. 
Any help is much appreciated. 

Thanks

Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

roramirez, 2017-01-03:
Hi @DRAZINED 

If you desactivate the modules Email Notification, MM Globe or other not included into default modules still happen?

DRAZINED, 2017-01-03:
If you deactivate the modules Email Notification, MM Globe or other not included into default modules still happen?: Yes, I ran it on its original version and froze up regardless.

roramirez, 2017-01-03:
Read this #570 
Sometime is problem with dependencies or power supply.

DRAZINED, 2017-01-03:
Safe to say a "known issue" then?
I don't want to trash it, it came out nice and already have a place to display it (heartbroken !)

MichMich, 2017-01-04:
My best guess is that it's your power supply. Had the same issue a few times.

DRAZINED, 2017-01-04:
Ok, so I remove the "compliments" from config file, RSS no longer static (scrolling w/o stopping), however not updating as desired. 

??

DRAZINED, 2017-01-06:
Solved!
Removed "compliments" and teak up RSS weather and RSS ".js" files.

closed 2017-01-06 by DRAZINED

roramirez, 2017-01-06:
That's interesting. Did you use compliments from remoteFile?

DRAZINED, 2017-01-06:
Default modules, non server configuration

DRAZINED, 2017-01-06:
My appreciation; too many modules maintaining RAM busy.