MagicMirrorOrg/MagicMirror #595: how many module can add with magic mirror at a time, bcz i want to add default and camera, music , voice command and

ManikandanET, 2017-01-04:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

morozgrafix, 2017-01-05:
It maybe worth opening a topic on forum.magicmirror.builders instead of filing an issue here

roramirez, 2017-01-05:
@morozgrafix  you're right.

The number of modules running is relative, The use of CPU or resource need for run for every these. So, is difficult determinated the "numbers" of module can be run. Other factor is the system you are using.

If you have a determinated issue or more details problem, some light, if dont run the modules -wrong configuration, crash the system, some working wrong, etc.. is possible help you.

So, you can go the [forum](https://forum.magicmirror.builders/) and post the question there, maybe other users has experience with your setup.

nhubbard, 2017-01-06:
#close @MichMich

closed 2017-01-10 by MichMich