MagicMirrorOrg/MagicMirror #602: [Proposal] config.js should be config.json

dolanmiu, 2017-01-06:
I am unconfortable with these lines in the config.js file: https://github.com/MichMich/MagicMirror/blob/master/config/config.js.sample#L81-L82

```js
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```

This will be resolved if the config file is correctly and aptly named `config.json`

Just to be clear, `.js` aka `JavaScript` files contain **logic**, `.json` files only contain data, and should be used for configuration.

roramirez, 2017-01-06:
Hi @dolanmiu 

The json format is not powerfull like a dict in Javascript. The design to use js file versus a json file have you sense. Please [read this](572), there are a lot explain why dont use a json file

dolanmiu, 2017-01-07:
you can also have javascript dict in `.json`

nhubbard, 2017-01-08:
@dolanmiu I also agree with @roramirez for another reason. Node likes to cache static JSON files, which makes Node forget that they have changed. This makes configuration very tough to fix if there is an issue. The only way that I have found to fix this is to export a JSON object from a JavaScript file. I get that it bothers you, but even if we did change it, this would require every module to have a refactor to make them work again. Therefore, we should not change this system.

cc @MichMich

bh, 2017-01-08:
:-1: for JSON because there is no support for comments unless you use https://hjson.org/.

dolanmiu, 2017-01-09:
@bh Why would you want comments in the config? Isn't the documentation enough?

For example, if I want to configure something, I would refer to the documentation, README, Wiki, or help pages. I would not look at a comment

roramirez, 2017-01-09:
@dolanmiu Hey, there a lot other technical stuff in pro for use a config like we use versus a JSON.  Asignate properties, functions, callback, "exotic content",  etc..

If you have a technical pro of JSON format can be send us a list about this, the "unconfortable" thing  is not a valid argument to me.

In the previous link I sended you before.  There a lot arguments why dont use a JSON format into the config file. These are the reasons. 

This topic was discuted many time before and the answer was the same.

MichMich, 2017-01-10:
I think other explained it. It will not be a json file due to the limitations of JSON. 

Feel free to reopen if you have additional questions.

closed 2017-01-10 by MichMich