MagicMirrorOrg/MagicMirror #611: Analog Clock face got scrambled

3vidar, 2017-01-11:
Hi,
After the update to V2.1.0 it got scrambled analog clock face. I resize analogSize to 150px and 300px and then it looks nice but after one day its again scrambled. Before the update i had 550px and it was fine. I disabled also all modules beside the analog clock and nothing changed, my analog face is still scrambled. I tested other faces and all are also scrambled. Someone know why this happend to me after updating MM? 

[http://i.imgur.com/wvjDlV5.png](http://i.imgur.com/wvjDlV5.png)

[Forum](https://forum.magicmirror.builders/topic/1365/clock-module-analog-face-got-scrambled/2)
[Forum](https://forum.magicmirror.builders/topic/318/analog-digital-clock-module/34)

MichMich, 2017-01-12:
It seems to work fine here ... could you share thes cononfig section of your clock module so I can try it?

3vidar, 2017-01-13:
Hi, thats my config part for the clock module:
`{
			module: 'clock',
			position: 'top_center',
				config: {

							displayType: 'analog',
							analogSize: '300px',
							analogFace: 'face-001', // options: 'none', 'simple', 'face-###' (where ### is 001 to 012 inclusive)
							analogPlacement: 'right', // options: 'top', 'bottom', 'left', 'right'
							analogShowDate: 'bottom', // options: false, 'top', or 'bottom'
							secondsColor: '#e20f0f',

									}
		},`
I updated my MM first with this:

`git fetch origin
git reset --hard origin
git clean -f and then git pull`

and then just replaced my Config and custom.css and after that my clock looks scrambled.

MichMich, 2017-01-14:
could you try it without your custom.css?

3vidar, 2017-01-14:
Hi,
I try it without my custom.css, but the clock looks the same. I try it also to fetch and reset again but its still the same.

MichMich, 2017-01-15:
Ah, indeed... got the same issue in Electron. Not in Safari.

MichMich, 2017-01-15:
I just pushed a fix to the `develop` branch. Goed you give it a try?

NB. Make sure you run `npm install` after pulling the develop branch, since it needs some new node modules.

3vidar, 2017-01-15:
You're a genius!!!! It works know like a charm. Thank you for your help and work!

MichMich, 2017-01-15:
Cool, thanks for testing.

closed 2017-01-15 by MichMich