MagicMirrorOrg/MagicMirror #618: Using Magic Mirror in Landscape, Weather forecast is too wide

NewRasNoClue, 2017-01-13:
I'm new to all of this.

I was able to set up and configure my magic mirror how i want it. I am using my screen in landscape. The only problem is that the default weather forecast table is too wide. I assume this is due to the landscape orientation width ratios. 

How do I make the weather forecast table more narrow/tight?

I left it positioned top_right.

nhubbard, 2017-01-14:
@NewRasNoClue Do you know how to use the custom CSS file? If so, I can tell you how to expand the top_right container.

MichMich, 2017-01-14:
Yeas, check out the css file like @nhubbard says.

NewRasNoClue, 2017-01-14:
I do not know how to use the custom CSS file for this. Can you reference an example, or explain? Thanks!

MichMich, 2017-01-15:
In the file: `css/custom.css` put something like:

````css
.weatherforecast {
    width:300px;
}
````

NewRasNoClue, 2017-01-15:
@MichMich that worked! thanks!
the table is narrow now, it is aligned toward the middle, rather than the right side. 
any tips on how to fix it so that it is aligned against the far right?
thanks!

NewRasNoClue, 2017-01-15:
I had a MMMstocks in the top right position. it was controlling the width of the top right region.
for now i removed the stocks and the region looks cleaner. 
but for future reference, how do i set the module to align to the right of the region rather than to the left?

MichMich, 2017-01-17:
There are many ways to accomplish that with each their own pros and cons. 
Take a look at some online css tutorials to know more about this topic, since it isn't MM specific.

I close this issue for now. Feel free to reopen.

closed 2017-01-17 by MichMich