MagicMirrorOrg/MagicMirror #637: Calendar Module Help

lucallmon, 2017-01-21:
This is more of a request for help as opposed to a needed fix for an issue. However, this may fit here anyway.

See the attached photos: what I'm trying to do is to get all of the calendars in my google calendar to show up under one single module window. I've gotten it to add two separate calendars by adding a second calendar module in the config.json file. As you can see, I have several sport team calendars that all come together into one window. Is there a way to link all of the gmail calendars into one single module window? 

[image]
[image]

nhubbard, 2017-01-21:
@lucallmon Just get all the ICS URLs for each calendar and add each calendar individually in the references. You can use multiple calendars. For icons, see here: https://fontawesome.io/icons

lucallmon, 2017-01-21:
Can you give me an example of the code to put in the config file?

MichMich, 2017-01-21:
Try something like this. :)

````
calendars: [
        {
            url: 'http://www.calendarlabs.com/templates/ical/US-Holidays.ics',
            symbol: 'calendar',
        },
        {
            url: 'http://second.calendar/url.ics',
            symbol: 'eye',
        },
    ],
````

For future support requests, please open a topic on http://forum.magicmirror.builders 

Thanks!

MichMich, 2017-01-21:
And to add: everything you need to know about the calendar can be found here: https://github.com/MichMich/MagicMirror/tree/master/modules/default/calendar

lucallmon, 2017-01-21:
I think I figured it out.

lucallmon, 2017-01-21:
Yep. Figured it out before you posted the help. thanks @MichMich. 

Also, I'm having trouble adding all my calendars from Gmail. Have you thought about adding support for RSS feed calendars?

the reason I'm having trouble is because my "other calendars" don't allow sharing. Any ideas on how to fix this? As you can see in the attached picture, I have a ton of other "non-ical" calendars that I'd like to add to the mirror. mostly sport stuff. Add HTML support for the calendar module also?

The one I really want is my Facebook birthdays and episode calendar, but I can't get it to allow sharing so I can link it to the mirror module. 

Perhaps add webcal? Thoughts?

MichMich, 2017-01-22:
I can't help you with this since I only use iCal, you might have better luck on: https://forum.magicmirror.builders/category/10/troubleshooting

Since this is a support ticket, I close it for now. Please continue ton the forum. Thanks! :)

closed 2017-01-22 by MichMich