MagicMirrorOrg/MagicMirror #639: Blank Black screen

ostfilinchen, 2017-01-22:
If I open the MirrorPage on my PC via http://raspberrypi:8080 i get a blank Black screen, But only with MS Edge. With a third Party Browser like Chrome or Opera i didn't have the issue. Also on the smartphonebrowser it works without issues.

roramirez, 2017-01-23:
Do have any log some inspector tool on MS Edge?

nhubbard, 2017-01-24:
@ostfilinchen MS Edge is known to not work. It has to do with Edge's rendering engine.

MichMich, 2017-01-24:
@nhubbard Is this something we should solve in your opinion?

nhubbard, 2017-01-27:
@MichMich I don't know. We write our code for Chrome, and if you have noticed, there are significant differences in Edge vs Chrome. It's like trying to run your code on Google Chrome when it only ran on Netscape Navigator.

MichMich, 2017-04-05:
Since Edge/Windows is a non supported platform, and this issue is inactive I'm closing this for now ...
If you feel this issue should remain open, please let me know.

closed 2017-04-05 by MichMich