MagicMirrorOrg/MagicMirror #662: TypeError: Cannot read property 'split' of undefined

Binog, 2017-01-28:
**Platform**: Raspberry pi3, 
**Node Version**: v6.9.4
**MagicMirror Version**: ppm mm -v = 3.10.10   /   MM 2.1.0
**Description**: First of all I get a full white page. the console says: 
´./mm.sh`
> magicmirror@2.1.0 start /home/pi/MagicMirror
> electron js/electron.js

Starting MagicMirror: v2.1.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
**WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Cannot read property 'split' of undefined**
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
App threw an error during load
TypeError: Cannot read property 'split' of undefined
    at loadModule (/home/pi/MagicMirror/js/app.js:71:24)
    at loadModules (/home/pi/MagicMirror/js/app.js:118:4)
    at /home/pi/MagicMirror/js/app.js:167:4
    at loadConfig (/home/pi/MagicMirror/js/app.js:59:5)
    at App.start (/home/pi/MagicMirror/js/app.js:155:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:101:6)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
Whoops! There was an uncaught exception...
TypeError: Cannot read property 'split' of undefined
    at loadModule (/home/pi/MagicMirror/js/app.js:71:24)
    at loadModules (/home/pi/MagicMirror/js/app.js:118:4)
    at /home/pi/MagicMirror/js/app.js:167:4
    at loadConfig (/home/pi/MagicMirror/js/app.js:59:5)
    at App.start (/home/pi/MagicMirror/js/app.js:155:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:101:6)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.

**Steps to Reproduce**: start via mm.sh
**Expected Results**: a normal magic mirror default screen,. never have seen it on my machine :-(

**Actual Results**: a blank white page

**Configuration**: 

`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"],

	language: 'de',
	timeFormat: 24,
	units: 'metric',

	modules: [
		{
			module: 'alert',
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: 'clock',
			position: 'top_left'
		},
		{
			module: 'calendar',
			header: 'Ferien',
			position: 'top_left',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'webcal://www.calendarlabs.com/**'
					}
				]
			}
		},
		{
			module: 'compliments',
			position: 'lower_third'
		},
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: '**',
				locationID: '**',  //ID from http://www.openweathermap.org
				appid: '**'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			config: {
				location: '**',
				locationID: '**',  //ID from http://www.openweathermap.org
				appid: '**'
			}
		},
//	has an error, too	{
//			module: 'mmm-systemtemperature',
//			position: 'top_center', // This can be any of the regions.
//			classes: 'small dimmed', // Add your own styling. Optional.
//			config: {
//			prependString: 'System Temp',
//			updateInterval: '5000',
//			animationSpeed: '0',
//			    // See 'Configuration options' for more information.
//			}
//		},
  
		{
				modules: [
				    {
				        module: 'helloworld',
				        position: 'bottom_bar', // This can be any of the regions.
				        config: {
				            // See 'Configuration options' for more information.
				            text: 'Hello world!'
				        }
				    }
				]
		},
		{
    module: 'MMM-DWD-WarnWeather',
    position: 'top_left',
    header: 'Wetterwarnungen',
    config: {
        region: '**',
        changeColor: true,
        interval: 10 * 60 * 1000, // every 10 minutes
        loadingText: 'Warnungen werden geladen...',
        noWarningText: 'Keine Warnungen'
} },
		{
         module: 'MMM-pihole-stats',
         position: 'top_left',
         config: {
         apiUrl: 'http://**/**/api.php',
         showSources: 'true',
         showSourceHostnameOnly: 'true',
         updateInterval:	'600000',
         retryDelay:	'2500',
                      // See 'Configuration options' for more information
         }
		},
		{
    module: "MMM-Fuel",
    position: "top_right",
    config: {
        api_key: "**",
        lat: **,
        lng: **,
        types: ["**"],
        open:	true,
        radius:	5,
        max:	5,
        height:	600,
        width:	600,
        colored:	true,
        shortenText:	false,
        showAddress:	false,
        showOpenOnly:	false,
        iconHeader:	true,
        rotate: true,
        rotateInterval:	60000,
        updateInterval:	900000,
			}
		},
		{
		    module: 'compliments',
			position: 'lower_third',
			config: {compliments: {
					day_sunny: [
         				   'Today is a sunny day',
        				    'It\'s a beautiful day'
       					 ],
       					 snow: [
       					     'Snowball battle!'
       					 ],
       					 rain: [
        				    'Don\'t forget your umbrella'
       					 ]
				},
		
		  module: 'voicecontrol',
            position: 'bottom_left',
            config: {
                models: [
                    {
                        keyword: "playMusic",   // keyword 
                        description: "Say 'Play Music' to start playing",
                        file: "playMusic.pmdl", // trained model file name
                        message: "PLAY_MUSIC"   // notification message that's broadcast in the MagicMirror app
                    },
                    {
                        keyword: "stopMusic",
                        description: "Say 'Stop Music' to stop playing",
                        file: "stopMusic.pmdl",
                        message: "STOP_MUSIC"
                    },
                ]
            }
		},
}
]
};
    

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
`

**Additional Notes**: Thanks for helping, cause I'm a novice. Jimmy

Binog, 2017-01-28:
If I uncomment current weather I get the black: create a config file-Screen... so something is wrong either with my config file or - preferrable AND - the current weather module.

nhubbard, 2017-01-28:
@Binog This seems to be a problem with the MMM-Fuel module. I can't help as much, but I am willing to help. You need to put an Array of Strings in the types section, like so:

```javascript
{
  types: ["diesel", "e5", "e10"]
}
```

closed 2017-01-29 by MichMich