MagicMirrorOrg/MagicMirror #664: MM.getModules returns undefied object if module is disabled in config file

KilianB, 2017-01-29:
If a module is disabled in the config file disabled: true
`MM.getModules()` will return an array with the length equal of all modules defined in the config files. If a module is marked as inactive (`disabled: true`) no class but rather an undefined object will be returned.

https://github.com/MichMich/MagicMirror/blame/8010e6220d4cc68e9f9e3c1029998cc47df0f193/js/main.js#L480-L483

Jopyth, 2017-01-29:
This is indeed a strange behavior in my opinion, but what I did in my remote control is: I switched to iterating via index to the `enumerate(fn () {...})` method, which fixes that.

However an array with `length = 7`, but the following elements still looks kind of strange to me. But apparently this is allowed in javascript
```
[0] = "A"
[1] = "B"
[3] = "D"
[5] = "F"
[6] = "G"
```

MichMich, 2017-04-05:
Closing due to inactivity. 
If you feel this issue should remain open, please let me know.

closed 2017-04-05 by MichMich