MagicMirrorOrg/MagicMirror #680: loader.js cannot console.log() or Log.log()

stanchanpsu, 2017-02-02:
I'm working on a PR to prevent custom.css from trying to load if it doesn't exist. While trying to debug my code, I noticed nothing written inside the immediately invoked function Loader can be logged to console via console.log or Log.log. Does anyone know why?

MichMich, 2017-02-02:
Could you share some code?

stanchanpsu, 2017-02-02:
I added this to line 23 of loader.js and it never prints out:

```
	var loadModules = function() {
		console.log("Yes I can.")
```

I also noticed I have never seen any of the other Log.log() appear in console.

roramirez, 2017-02-03:
Seems you are out-date on git. What is the last commit id?

stanchanpsu, 2017-02-03:
What makes you think I am out of date? I git pulled on my contribution branch just now.

Git commit: commit fd738e5c8b6938d2097d91eac836e4b025c463d1

Logs are still not showing form loader.js

morozgrafix, 2017-02-03:
I wasn't able to reproduce it on the latest version of the master and develop branches.
Added to line 23 of the `loader.js`:
```
var loadModules = function() {
	console.log("Mirror mirror on the wall!");
```

4th line in the Chrome console.log while running `node serveronly` had my expected string:
```
Initializing MagicMirror.
Loading core translation file: translations/en.json translator.js:182 
Loading core translation fallback file: translations/en.json translator.js:204 
Mirror mirror on the wall! loader.js:23 
Load script: modules/default/alert//alert.js loader.js:183 
Module registered: alert module.js:415 
```

roramirez, 2017-02-03:
@stanchanpsu I mentioned you because if dont have you this fix https://github.com/MichMich/MagicMirror/commit/ee0418e7197547765d9427f6ed6f9531d85f9a32 the Logs will be not working.

stanchanpsu, 2017-02-03:
@morozgrafix, I will try to reproduce with `node serveronly`. I am unable to see any logs form loader.js from within the terminal when I run `npm start` and use the electron frontend. Could you try to reproduce with `npm start`?

stanchanpsu, 2017-02-03:
@roramirez I will make sure to pull develop again and check for that hotfix.

stanchanpsu, 2017-02-04:
I'm going to close this because I realized that only in `serveronly` mode does it log everything. I will open a new issue for help with checking if a file exists before loading.

closed 2017-02-04 by stanchanpsu

stanchanpsu, 2017-02-04:
Here is the forum post if you guys are interested in helping me out: https://forum.magicmirror.builders/topic/1646/check-to-see-if-file-ex-custom-css-exists-before-trying-to-load-it