MagicMirrorOrg/MagicMirror #682: Analog clock not scaling correct the faces

olovskos, 2017-02-03:
I don't know why and I`m not interested in digging deeper into but if you use the analog clock, you cannot resize it. It just shows multiple of the face selected

morozgrafix, 2017-02-03:
@olovakos can you please provide relevant snippets of your `config.js`? Thanks

olovskos, 2017-02-03:
Yes of course I can :)       

{
        module: "clock",
        position: "top_left", 
        config: {
            displayType: 'analog',
            analogSize: '300px',
            analogFace: 'face-002'
        }
}

morozgrafix, 2017-02-03:
I wasn't able to reproduce it in Chrome or Electron on my mac on both latest version of master and develop branches. I don't have access to RasPi until later to test it out. Here is what I get when I use settings that you've provided:

[image]

olovskos, 2017-02-03:
This is how it looks like
[image]

BUT
I changed the size of the fonts by changeing the custom.css regarding this link  
[https://forum.magicmirror.builders/topic/74/adjust-modules-size-to-match-each-other-weather-calendar/2](url)

And now it looks fine but this can't be the solution of the problem. 

I`m using a RPI 3 with the newest raspbian.

morozgrafix, 2017-02-03:
Seems that few people on the forum are [having issues with it](https://forum.magicmirror.builders/search?term=analog%20clock&in=titlesposts) scaling correctly

roramirez, 2017-02-04:
Maybe @KirAsh4 can help with this. He introduced the feature for analog clock ;)

MichMich, 2017-02-07:
@olovskos what branch/version are you running? I used a fix for this a while ago.

KirAsh4, 2017-02-09:
Since released, I have not doing any scaling tests with subsequent updates to electron, but it appears to be broken now. I'll have to go dig into the CSS a bit and figure out why it's failing.

MichMich, 2017-02-09:
@KirAsh4, are you testing the develop branch, or the master branch?

KirAsh4, 2017-02-09:
I like living on the edge.
develop branch.

MichMich, 2017-02-09:
Then it should be solved ... it isn't?

MichMich, 2017-04-05:
Closing due to inactivity.

closed 2017-04-05 by MichMich

KirAsh4, 2017-04-05:
I don't think this was an actual issue after it was fixed (by you). I haven't had any problems with it.

MichMich, 2017-04-05:
Good to hear! Thanks.