MagicMirrorOrg/MagicMirror #683: Can't fetch URL's

Smartiesbehindproxy, 2017-02-03:
I'm currently working on the MagicMirror at the School for a Project.
Everything worked fine at first, but after a few times it stopped working i'm getting two errors : 
```
{ Error: connect EHOSTUNREACH 151.101.193.164:80
    at Object.exports._errnoException (util.js:1026:11)
    at exports._exceptionWithHostPort (util.js:1049:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1081:14)
  code: 'EHOSTUNREACH',
  errno: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '151.101.193.164',
  port: 80 }
``` 
The School acts behind a proxy, but I export the proxy already. 
And i'm  also able to ping the IP address shown in the error. 

next Error : 
```
{ Error: socket hang up
    at createHangUpError (_http_client.js:252:15)
    at Socket.socketCloseListener (_http_client.js:284:23)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at TCP._handle.close [as _onclose] (net.js:493:12) code: 'ECONNRESET' }
```
So I assume, because im not able to fetch the URL i get a socket hang up. Not sure.

I already reinstalled it, even the OS.
Getting the same error.
At the first time starting ,after changing Calendar etc., it worked fine but after the third run it stopped working.

Any suggestions on how to fix it?
Is the proxy maybe blocking the requests/answers?

morozgrafix, 2017-02-03:
It's very possible that your request isn't going through the proxy. What happens when you run following command from terminal of your RasPi?

`curl -v -L 151.101.193.164`

Smartiesbehindproxy, 2017-02-03:
@morozgrafix I will try that on Monday and let you know about the result, thanks.

Smartiesbehindproxy, 2017-02-06:
@morozgrafix 
I used your command and got the following : 

```
root@raspberrypi:/home/pi# curl -v -L 151.101.193.164
* Rebuilt URL to: 151.101.193.164/
* Hostname was NOT found in DNS cache
*   Trying 151.101.193.164...
* connect to 151.101.193.164 port 80 failed: No route to host
* Failed to connect to 151.101.193.164 port 80: No route to host
* Closing connection 0
curl: (7) Failed to connect to 151.101.193.164 port 80: No route to host
```
and then i tried to use the proxy by `--proxy`.
so : `curl --proxy http://proxyip:port -v -L 151.101.193.164 `
and then i got : 
```
* Rebuilt URL to: 151.101.193.164/
* Hostname was NOT found in DNS cache
*   Trying proxyip...
* Connected to proxyip (proxyip) port (#0)
> GET HTTP://151.101.193.164/ HTTP/1.1
> User-Agent: curl/7.38.0
> Host: 151.101.193.164
> Accept: */*
> Proxy-Connection: Keep-Alive
> 
< HTTP/1.1 500 Internal Server Error
* Server Varnish is not blacklisted
< Server: Varnish
< Retry-After: 0
< Content-Type: text/html
< Cache-Control: private, no-cache
< Content-Length: 209
< Accept-Ranges: bytes
< Date: Mon, 06 Feb 2017 07:24:05 GMT
< X-Cache: MISS from proxy
< X-Cache-Lookup: MISS from proxy:3128
< Via: 1.1 varnish, 1.1 proxy (squid/3.3.8)
< Connection: keep-alive
< 

<html>
<head>
<title>Fastly error: unknown domain 151.101.193.164</title>
</head>
<body>
* Connection #0 to host proxyip left intact
Fastly error: unknown domain: 151.101.193.164. Please check that this domain has been added to a service.</body></html> 
```

morozgrafix, 2017-02-06:
It seems that your proxy is not configured to allow connection and not the issue with the MagicMirror code

Smartiesbehindproxy, 2017-02-06:
So i have no chance to avoid this? Maybe get information, store them somewhere and then implement them somehow?

morozgrafix, 2017-02-06:
The issue that you seem to experience has to do with proxy not allowing http traffic from your machine to that specific host. It's not really an issue with MagicMirror code itself. Talk to your network administrator and see why your proxy restricts that access and if there is anything that can be done.

Smartiesbehindproxy, 2017-02-06:
Thanks for the help, I talked to the network administrator and we figured out that he can't do anything, because our school is configured  by a company from the state. But I managed to avoid the problem by using VPN with another RasPi.
And it works all fine, they will talk to the company and solve this problem.

closed 2017-02-06 by Smartiesbehindproxy