MagicMirrorOrg/MagicMirror #684: Unable to start electron with dev option on develop branch

morozgrafix, 2017-02-03:
@roramirez I believe that after PR #652 `npm start dev` is no longer working on develop branch, because `run-start.sh` doesn't process `dev` parameter.

qistoph, 2017-02-03:
Suggestion, when it's fixed. Add a test case to verify it too ;-)

roramirez, 2017-02-03:
I pushed #685  to resolv this case.

@morozgrafix @qistoph do you have any approach how test this case?

qistoph, 2017-02-05:
I was thinking something like this:
https://github.com/qistoph/MagicMirror/commit/42f22119f2c2564314279b640ca0bd6884e31ed5

However, just found out that this does not include run-start.sh in the e2e-test...

roramirez, 2017-02-05:
Yes, this is work in  layer inside the electron not the script run-start.sh. I seems good this test to me, later we can include a new for run-start.sh

morozgrafix, 2017-02-11:
I think we can close this one after PR #711 goes through.

closed 2017-04-05 by MichMich