MagicMirrorOrg/MagicMirror #691: Load fonts and vendor libraries via bower.

Bruce17, 2017-02-06:
Hi,

I noticed you included a lot of fonts and vendor libraries directly into your project. How about loading them via [bower](https://bower.io/)?

paviro, 2017-02-06:
npm would make more sense wouldn't it? Most are available there too and it needs to be use anyway for the node dependencies.

Bruce17, 2017-02-06:
Yeah sure can use npm too. Personally I like to use bower for frontend stuff only to separate them a little bit.

If you like, I can try to remove some libraries and add them via npm (or bower, what you prefer).

roramirez, 2017-02-06:
There a related discussion about the vendor directory #592.

Bower is greats for keep externals libs into the a project without tracking the files in version control system.

Why not try a send a Pull Request about this?

MichMich, 2017-02-07:
If we gonna use a package manager, let's stick to NPM. I don't want to overcomplicate things.

The reason why some things are in the vendor folder is because they are modified versions because of a bug in the current version. This, of course, could use a better solution.

closed 2017-02-07 by Bruce17