MagicMirrorOrg/MagicMirror #707: Can´t see openweather module currentweather and forecast

Phoenxi131178, 2017-02-09:
Hello Michael
At first thx for the MagicMirror Software. For me it is the first time to check the Raspberry and Linux.
I installed the Ras3 with raspbian jessie and then i installed your Magic Mirror with 
curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash
I used complete your manual.
I got an account from openweather.org and filled my appid in the config.js and the module currentweather and weatherforecast.
I checked everything twice but now i have no idea.
I hope everyone can help me. Maybe i have forgotten somthing to install ??

Greetings from Germany 

Jan

fewieden, 2017-02-09:
Hey Jan, do you get any errors in the terminal or electron console (npm start dev)? Probably some syntax errors in your config, you can check your config.js for example on jshint.com

MichMich, 2017-02-10:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-02-10 by MichMich

Phoenxi131178, 2017-02-10:
Thank you for your help !!! I got it.
One wrong Nummer in the APPID. npm start dev showed me what was wrong.

Greeting Jan