MagicMirrorOrg/MagicMirror #709: [TRACKING ISSUE] Engine.io, and by extent, Socket.io, has a vulnerability

nhubbard, 2017-02-10:
**Platform**: Affects all platforms.

**Node Version**: Affects all versions.

**MagicMirror Version**: v2

**Description**: Engine.io, the stack behind Socket.io, has a vulnerability caused by an out-of-date version of the `ws` module. Tracking issue in socketio/engine.io is available [here](https://github.com/socketio/engine.io/issues/479)

roramirez, 2017-02-10:
I can see. Its in the planning to fix soon.

I think is need it before release a new version add this fix from the new version of `ws` module. 

By other hand, If is release soon the patch of this. How w'll process to apply a patch on master branch? 

[I've mentioned before](https://github.com/MichMich/MagicMirror/pull/591#discussion_r94590779) about release label, we using [Semantic Versioning](http://semver.org/), What be the next release version name 2.1.0.1 or 2.1.1 if include only this patch?

nhubbard, 2017-02-11:
@roramirez When they release a newer version with an updated `ws` module, we can release it with the next new version.

roramirez, 2017-02-11:
@nhubbard My question is more related, what will happened in the future if we need add only one patch on master branch?, What version name will be?

morozgrafix, 2017-02-11:
If you patch current version with a backwards compatible bug fix or patch, v2.1.0 would become v2.1.1 if I'm not mistaken. As far as I recall Semantic Versioning follows `vMajor.Minor.Patch` scheme.

roramirez, 2017-02-11:
Yes, I understand the version semantic that way. The main point is next planning name version is **2.1.1** I think should be **2.2.0**

nhubbard, 2017-02-11:
@morozgrafix Spot on.

MichMich, 2017-02-11:
2.2.0 will be used whenever there is a breaking change in the API. (like 2.1.0 was)...

2.1.1 will include this security patch.

nhubbard, 2017-02-19:
@MichMich They released version 2.0.2 of Engine.io... but I don't see any new version of Socket.io :cry:

closed 2017-02-19 by nhubbard

roramirez, 2017-02-19:
This  https://github.com/socketio/socket.io/releases/tag/1.7.3 ?

nhubbard, 2017-02-19:
@roramirez Yup. 1.7.3 bumped engine.io-client to 1.8.3, which contained the bugfix.

roramirez, 2017-02-19:
So we should be moved this in package.json. I pushed  a PR.