MagicMirrorOrg/MagicMirror #710: Enabling private messages between modules

fewieden, 2017-02-10:
I added the possibility to share private messages between modules.

What do you think about it? https://github.com/fewieden/MagicMirror/commit/c2d9656e4ec2461a8cd5f6c8f5f59b8c98b51316

roramirez, 2017-02-11:
What is the context where can use?. I think can be more clean create new methods to send/receive private notifications instead of same functions with new parameters.

fewieden, 2017-02-11:
A lot of modules listen on some generic broadcasts, but i want to inform just one module about it.

I thought adding a new parameter is the best, because everything will work as it was before and if you want to send it private you just add an receiver to it

MichMich, 2017-02-11:
This is how most notifications systems work. You listen to a specific notification and if it is any interest to you, you do something with it. We don't create any real overhead by doing it this way.

By making notifications semi-private (because it's JS, it will not be secure anyway). you'll only limit the possibilities for other modules. 

I dont see any good reason to enable private messages.

closed 2017-02-12 by fewieden