MagicMirrorOrg/MagicMirror #717: Current Weather Module 0° Bug

likeadoc, 2017-02-12:
**Platform**: Rpi3

**MagicMirror Version**:  JavaScript version (v2).

**Description**: If the temperature is exactly 0.0° the currentweather module is suspended due to the poor checking mechanics of the initial if statement in processWeather (!data.main.temp).
It would be better to check for the pressure (data.main.pressure) instead of the temperature.

roramirez, 2017-02-13:
This should be fix in commit 93c07b2b1e2e4218d17ceead43b63509b6beb7e0 

What version and branch are you using?

The previous change is in the develop branch. Are you using from the master branch?

likeadoc, 2017-02-13:
sorry for your inconvenience ! I´m on the master branch - so yeah:D

closed 2017-02-13 by likeadoc