MagicMirrorOrg/MagicMirror #723: It is not possible to start MM with latest node/npm versions (clean install)

bh, 2017-02-17:
**Steps to Reproduce**
```
cd ~/MagicMirror
rm -rf ./node_modules
git pull
npm install
npm start
```
**Platform**: Raspberry Pi 2 (Arch Linux ARM)
**Node Version**: 7.5.0
**NPM Version**: 4.2.0

**MagicMirror Version**: devel branch (11c0221f8172e9043473d4e42ae5c0cd994b9d59)

**Issue**:
It is not possible to start the electron magic with latest node/npm versions:

`$ npm start`

> magicmirror@2.1.1 start /home/mm/MagicMirror
> sh run-start.sh
> 
> /home/mm/MagicMirror/node_modules/electron/index.js:9
>   throw new Error('Electron failed to install correctly, please delete node_modules/' + path.basename(__dirname) + ' and try installing again')
>   ^
> 
> Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
>     at Object.<anonymous> (/home/mm/MagicMirror/node_modules/electron/index.js:9:9)
>     at Module._compile (module.js:571:32)
>     at Object.Module._extensions..js (module.js:580:10)
>     at Module.load (module.js:488:32)
>     at tryModuleLoad (module.js:447:12)
>     at Function.Module._load (module.js:439:3)
>     at Module.require (module.js:498:17)
>     at require (internal/module.js:20:19)
>     at Object.<anonymous> (/home/mm/MagicMirror/node_modules/electron/cli.js:3:16)
>     at Module._compile (module.js:571:32)
> 
> npm ERR! Linux 4.4.48-2-ARCH
> npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
> npm ERR! node v7.5.0
> npm ERR! npm  v4.2.0
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! magicmirror@2.1.1 start: `sh run-start.sh`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the magicmirror@2.1.1 start script 'sh run-start.sh'.
> npm ERR! Make sure you have the latest version of node.js and npm installed.
> npm ERR! If you do, this is most likely a problem with the magicmirror package,
> npm ERR! not with npm itself.
> npm ERR! Tell the author that this fails on your system:
> npm ERR!     sh run-start.sh
> npm ERR! You can get information on how to open an issue for this project with:
> npm ERR!     npm bugs magicmirror
> npm ERR! Or if that isn't available, you can get their info via:
> npm ERR!     npm owner ls magicmirror
> npm ERR! There is likely additional logging output above.
> 
> npm ERR! Please include the following file with any support request:
> npm ERR!     /home/mm/.npm/_logs/2017-02-17T20_32_08_740Z-debug.log
>

roramirez, 2017-02-17:
Seems a Electron issue not of the MagicMirror. 

I can see the last Electron support is Node 7.4.0
[image]

Did you try ask there?

bh, 2017-02-17:
Ohhh, than the Readme.rd is not really correct ("install the latest version of NodeJS") ;-)

roramirez, 2017-02-20:
I pushed a PL #727 to include Node7 in Travis CI. The tests are passing with Node v7.5.0
https://travis-ci.org/roramirez/MagicMirror/jobs/203279630

What electron version version did install?
` ./node_modules/electron-prebuilt/dist/electron -v `

bh, 2017-02-22:
OMG, all work fine now. I think it was not an Electron issue but much more a config issue. The chromium got a 403 Forbidden when requesting localhost:8080.  Here is my config patch:
```
[mm@mm MagicMirror]$ diff -u config/config.js.back config/config.js
--- config/config.js.back       2017-02-19 10:23:20.742176813 +0000
+++ config/config.js    2017-02-22 20:41:44.906738957 +0000
@@ -7,9 +7,10 @@
                height: 1080
        },

-       address: "10.1.1.192",
+//     address: "10.1.1.192",
         port: 8080,
        ipWhitelist: [
+               "::1",
                "10.1.1.25",
                "::ffff:10.1.1.25",
        ],
```

bh, 2017-02-22:
FYI: Node 7.6.0 works fine ;-)

closed 2017-02-22 by bh

MichMich, 2017-02-23:
Thanks for reporting @bh, greatly appreciated!