MagicMirrorOrg/MagicMirror #724: install problems for several times,..

Binog, 2017-02-19:
**Platform**: Raspberry Pi 3

**Node Version**: v4.3.0

**MagicMirror Version**: Guess the the newer JavaScript version (v2).

**Description**: I followed this guide after having it a third or fourth time installed:
`https://forum.magicmirror.builders/topic/236/complete-setup-tutorial/6`

**Steps to Reproduce**: Auto installation like in the guide above.

**Expected Results**: a clean install and the default screen

**Actual Results**: 
in the automatic installation at first I see this:

```
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g
npm WARN prefer global colorguard@1.2.0 should be installed with -g

> electron@1.4.15 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js
...
├─┬ UNMET PEER DEPENDENCY grunt@1.0.1
...
npm WARN grunt-yamllint@0.2.0 requires a peer of grunt@~0.4.0 but none was installed.

```
 Then I  install the above all to gather without errors..

Then I installed: 

```
sudo npm install -g grunt-yamllint
/usr/lib
├── UNMET PEER DEPENDENCY grunt@~0.4.0
└─┬ grunt-yamllint@0.2.0 
  ├── async@0.9.2 
  ├─┬ chalk@0.5.1 
  │ ├── ansi-styles@1.1.0 
  │ ├── escape-string-regexp@1.0.5 
  │ ├─┬ has-ansi@0.1.0 
  │ │ └── ansi-regex@0.2.1 
  │ ├── strip-ansi@0.3.0 
  │ └── supports-color@0.2.0 
  └─┬ js-yaml@3.8.1 
    ├─┬ argparse@1.0.9 
    │ └── sprintf-js@1.0.3 
    └── esprima@3.1.3 

npm WARN grunt-yamllint@0.2.0 requires a peer of grunt@~0.4.0 but none was installed.
```
and again

```
sudo npm install -g grunt@~0.4.0
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
- ansi-regex@2.1.1 node_modules/grunt/node_modules/ansi-regex
- ansi-styles@2.2.1 node_modules/grunt/node_modules/ansi-styles
- array-find-index@1.0.2 node_modules/grunt/node_modules/array-find-index
- balanced-match@0.4.2 node_modules/grunt/node_modules/balanced-match
- builtin-modules@1.1.1 node_modules/grunt/node_modules/builtin-modules
- camelcase@2.1.1 node_modules/grunt/node_modules/camelcase
- concat-map@0.0.1 node_modules/grunt/node_modules/concat-map
- brace-expansion@1.1.6 node_modules/grunt/node_modules/brace-expansion
[... 196 lines cut ...]
22 verbose stack     at emitTwo (events.js:106:13)
22 verbose stack     at ChildProcess.emit (events.js:191:7)
22 verbose stack     at maybeClose (internal/child_process.js:877:16)
22 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
23 verbose cwd /home/pi/MagicMirror
24 error Linux 4.4.49-v7+
25 error argv "/usr/bin/nodejs" "/usr/bin/npm" "bugs" "magicmirror"
26 error node v6.9.5
27 error npm  v4.3.0
28 error code 3
29 error Command failed: xdg-open https://github.com/MichMich/MagicMirror/issues
29 error /usr/bin/xdg-open: 1: eval: www-browser: not found
29 error /usr/bin/xdg-open: 1: eval: links2: not found
29 error /usr/bin/xdg-open: 1: eval: elinks: not found
29 error /usr/bin/xdg-open: 1: eval: links: not found
29 error /usr/bin/xdg-open: 1: eval: lynx: not found
29 error /usr/bin/xdg-open: 1: eval: w3m: not found
29 error xdg-open: no method available for opening 'https://github.com/MichMich/MagicMirror/issues'
30 error If you need help, you may report this error at:
30 error     <https://github.com/npm/npm/issues>
31 verbose exit [ 1, true ]

```

What Am I doing wrong??

nhubbard, 2017-02-19:
@Binog You can ignore the unmet peer dependency; it is a result of our development process. Your mirror should work with any of the errors above.

Binog, 2017-02-20:
Thanks for replying. Well I installed it as mentioned several times, cause the mirror quits unexpectedly after several minutes. now it seems to work. except that some modules do not work,. which worked in the first place. Wonderground, airquality, pinhole-stats,.. could those errors above be the cause. didn't change the config...

MichMich, 2017-02-20:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-02-20 by MichMich

Binog, 2017-02-20:
well, I still don't know weither I should install the npm packages (grant,..) manually, or if I can totally ignore this. If so, I think there shouldn't be an error message during installation,.. even if this is a result of you development process.. it really confuses me (and I guess others,..)