MagicMirrorOrg/MagicMirror #732: Umlauts in Compliments

autt87, 2017-02-20:
I am new to the MagicMirror (got it set up using default configs 4 days ago) and have found something I need help with that hasn't been talked about anywhere I can find.

I need to be able to write Umlauts äöüß in the compliments module. I have seen it asked in another issue regarding the newsfeed and even there when I try to use it it causes a 'crash' where MagicMirror doesn't recognize the config.js file. Any help would be appreciated, this is the icing on the cake in my project; getting umlauts in all the modules.

Thanks in advance for your guidance!

fewieden, 2017-02-20:
I don't have issues with those umlauts, is your config.js file utf-8 encoded? With the wrong encoding the config is still valid and should show up but not displaying them correctly.

I guess that the issue is based on quotation marks (only ```"``` and ```'``` are valid), wrong brackets (did you close all brackets that got opened) or comma (forgot about seperating with comma). You can check your config file on jshint.com, if this doesn't help I suggest to open a thread on the forum and add a snippet of your config

autt87, 2017-02-20:
Where in the config.js so I need to put the UTF-8 in?

Jopyth, 2017-02-21:
If it is really an encoding thing, you can always use HTML escapes, for example `&Auml;` becomes `Ä`. See a full list, e.g. [this](http://designerzone.de/html-umlaute-und-sonderzeichen/).

autt87, 2017-02-21:
I would like to have the ability to put in a encoding value somewhere in the config files so that it does the rest of the work for me. I just can't seem to get the placement right or when I follow the other examples the config file breaks. If someone has the encoding working on their mirror i'd simply like to now where they put the encoding argument in their modules. Examples or snippets/screenshots would be most helpful :)

autt87, 2017-02-21:
This is what I am looking at now :/

[image]

fewieden, 2017-02-21:
You don't put that in the config, you have to save the file as utf8 encoded. But as I said I guess that is not the problem that the mirror doesn't start, this is only affecting not disoplaying characters correctly.

autt87, 2017-02-21:
And yet, if I look at this: 
[image]

I get:
[image]

KX900, 2017-02-26:
I have the same problem with umlauts in Compliments.
MM2 works, but it doesn't display the Compliments when i type in:
encoding: "ISO-8859-1"
This was the solution for the same problem in newsfeed.
So what to do in compliments?

roramirez, 2017-03-22:
@KX900 Can you share your config?

autt87, 2017-03-22:
@roramirez So I found a work around that worked for me. I took the compliments.js file and scp'd it to my computer. I then used different editors to view the file. All was looking about the same so I took my editor of choice: Atom.app and re-wrote all umlauts the sentences the way I wanted it to look. 
Using Terminal I nano'd the .js file on the pi and removed the morning, afternoon, and evening fields and copied that from atom.
I stopped the magic mirror process and viewed the file using Terminal in Jessie. There everything finally looked GOOD! The umlauts and sharp S where just how they looked in Atom. Started up the Magic Mirror again and everything was working. I hope this helps you guys out. (My OS system language is US should you want to try that out as well).

closed 2017-09-06 by MichMich