MagicMirrorOrg/MagicMirror #735: ERROR! Could not find main module js file.

Binog, 2017-02-23:
**Platform**: (Raspberry Pi 3).

**Node Version**: 

**MagicMirror Version**: 201).

**Description**: By starting MM I get the following dialogue
```
(node:14713) DeprecationWarning: sys is deprecated. Use util instead.
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE :::8080
    at Object.exports._errnoException (util.js:1026:11)
    at exports._exceptionWithHostPort (util.js:1049:20)
    at Server._listen2 (net.js:1253:14)
    at listen (net.js:1289:10)
    at net.js:1399:9
    at _combinedTickCallback (internal/process/next_tick.js:77:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
    at Module.runMain (module.js:592:11)
    at run (bootstrap_node.js:402:7)
    at startup (bootstrap_node.js:157:9)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
ERROR! Could not find main module js file.
ERROR! Could not find main module js file.
```
**Steps to Reproduce**: Actually it ofter works till the ERROR! Could not... message. seems kind of unstable. often crashes,.. Any idea?

**Expected Results**: No Error at all, actually,....

**Actual Results**: White or black full screen...no information on it.

**Configuration**: 
```

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1", "xxxx/24", "::ffff:xxxx/24"],
	language: 'de',
	timeFormat: 24,
	units: 'metric',

	modules: [
		{
			module: 'alert',
		},
		{
			module: 'updatenotification',
			position: 'top_bar'
		},
		{
			module: 'clock',
			position: 'top_left'
		},
		{
            		module: 'calendar_monthly',
          		position: 'top_left',
           		config: {
        	      		showHeader: false,
                  		cssStyle: 'block'
               			},
    	},
		{
			module: 'calendar',
			header: 'Was .. wann ..?',
			position: 'top_left',
			config: {
					maximumEntries: 4,
					maximumNumberOfDays: 7,
					calendars: [
					{
						symbol: 'xxx',
						url: 'webcal://xxxx',
					}
				]
			}
		},
		{
			module: 'calendar',
			header: 'Feiertrage',
			position: 'top_left',
			config: {
					maximumEntries: 4,
					maximumNumberOfDays: 30,
					calendars: [
					{
						symbol: 'Feiertage',
						url: 'xxx',
					}
				]
			}
		},	

		{
			module: 'compliments',
			position: 'top_center'
		},
		{
  			module: 'MMM-WunderGround',
   			position: 'top_right',
    			config: {
       			apikey: 'xxx', // private; don't share!
        		pws: 'pws:xxx', //culemborg
        		hourly: '1',
        		fctext: '1',
        		fcdaycount: "4",
        		fcdaystart: "0",
        		hourlyinterval: "3",
        		hourlycount: "2",
        		alerttime: 10000,
        		alerttruncatestring: "english:",
			roundTmpDecs: 1,
			UseCardinals: 0,
			layout: "horizontal",
			sysstat: 0
    			}
		},
		{
			module: 'MMM-DWD-WarnWeather',
			header: 'Wetterwarnungen',
			position: 'top_right',
			config: {
				region: 'xxx',
				changeColor: true,
				interval: 600000,
				loadingText: 'Warnungen werden geladen...',
				noWarningText: 'Alles ruhig',
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_center',
			config: {
				feeds: [
					{
						title: "xx",
						url: "xxx"
					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				Interval: 100000,
				updateInterval:10000,
				animationSpeed:1000,
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_center',
			config: {
				feeds: [
					{
						title: "xx",
						url: "xxx"
					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				Interval: 100000,
				updateInterval:9000,
				animationSpeed:1000,
			}
		},
		{
			module: 'MMM-Fuel',
			position: 'top_right',
			config: {
				api_key: "xxx",
				lat: xxx,
				lng: xxx,
				types: [
					"e10"
				],
				open: true,
				radius: 2,
				max: 4,
				height: 600,
				width: 600,
				colored: true,
				shortenText: false,
				showAddress: false,
				showOpenOnly: true,
				iconHeader: false,
				rotate: false,
				rotateInterval: 60000,
				updateInterval: 900000,
				}
		},

		{
			module: 'mmm-systemtemperature',
			position: 'top_center',
			config: {},
			classes: 'small dimmed'
		},			
		{
			module: 'MMM-Globe',
			position: 'middle_center',
			config: {
				style: 'europeDiscNat',
				imageSize: 600,
				ownImagePath: "",
				updateInterval: 900000
			}
		},	
		{
      	 	module: 'MMM-WatchDog',
      	 	config: {
      	    	interval: 5,
      	    	timeout: 10,  // See 'Configuration options' for more information.
        		}
    	},	
    	{
			module: 'MMM-Admin-Interface'
		},
		{
			module: 'MMM-Remote-Control'
		},
		{
			module: 'MMM-pihole-stats',
			position: 'bottom_left',
			config: {
				apiURL: 'xxx',
			}
		},
		{
 		    module: 'MMM-voice',
    		position: 'bottom_left',
    		config: {
     			microphone: 1,
     			keyword: 'MAGIC MIRROR',
     			timeout: 15,
    		 }
 		},
		{
			module: 'MMM-NetworkScanner',
			header: 'Wer ist zu Hause',
			position: 'top_left',
			config: {
				devices: [
					{ macAddress: "xxx", name: "xx", icon: "xx"},
					{ macAddress: "xxx", name: "xx", icon: "xx"},
					{ macAddress: "xxx", name: "xx", icon: "xx"},
				],
				showUnknown: false,
				showOffline: false,
				keepAlive: 3000,
				updateInterval: 3600,
				residents: "mobile",

				}
		},	
		
		
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

```

yesyesuk, 2017-02-23:
"Error: listen EADDRINUSE :::8080"

some other application seem to already use port 8080.

Binog, 2017-02-23:
Yeah I thought so, how can I find out which one it is? I didn't install any other webserver on purpose.

roramirez, 2017-02-23:
Did you installed by installer script or manually?

MichMich, 2017-02-23:
Maybe, try rebooting. That's the easy way to solve it. ;)

Binog, 2017-02-23:
Yes I did couple of time,. rebooting. Did installed automatically. Found one reason: two crontab files,.. deleted one,. but now I do have 3 left,..don't know why...
[image]

nhubbard, 2017-02-26:
@Binog The processes you're showing are the Chromium and Electron processes. You shouldn't normally need to kill them. It's okay if there is more than one of them.

Binog, 2017-02-26:
Thank You for replying. I thought so, but wasn't sure, cause had the port 8080 is in use by another instance error. And the hud is still crashing. although,. I ruled out a lot of errors. If I start by cronjob pm2 then it crashes,. if I do this manually it doesn't,.. don't understand that,..

nhubbard, 2017-02-27:
@Binog Can you post the output of this command?

```bash
netstat -tulpn | grep 8080
```

mkings007, 2017-03-14:
I am also having same issue, please help...
[image]

MichMich, 2017-04-05:
Most of these issues are discussed on the forum. I'm closing this issue for now. 
If you feel this issue should remain open, please let me know.

closed 2017-04-05 by MichMich

PooLP, 2017-05-17:
Hi,
I have the same error.
I change the port and reboot the application too often ... it is not very practical to code a module.

typhonn353, 2018-06-10:
Hi, 
I have the same Problem electron.js use the 8080 port.
@nhubbard if I use your command I get 

(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      1120/electron.js