MagicMirrorOrg/MagicMirror #748: Add option for digest authentication in default calendar module

deg0nz, 2017-03-05:
This isn't an issue, but more an enhancement.

The `ical.js` used by default calendar module supports digest authentication by default, since it uses the `request` module internally. It's just the [sendImmediately option in calendarFetcher.js](https://github.com/MichMich/MagicMirror/blob/8010e6220d4cc68e9f9e3c1029998cc47df0f193/modules/default/calendar/calendarfetcher.js#L39) that needs to be set to `true`, if you want to use digest.
It's written in the [HTTP authentication section of `request`](https://github.com/request/request/tree/fa48e67578a3c43f83c9e4e9339440e8dbbcf6f5#http-authentication):

> Digest authentication is supported, but it only works with sendImmediately set to false; otherwise request will send basic authentication on the initial request, which will probably cause the request to fail.

I already tried this with my baikal server and it's working. 

Since I don't want to change this setting every time the mirror is updated, I'd like to implement this feature for the calendar module and send a pull request so it can be configured in the config file.

Is this OK for you, @MichMich ?

MichMich, 2017-03-05:
Feel free to send a PR to the develop branch! :)

deg0nz, 2017-03-05:
`request` also supports OAuth/Bearer authentication. This could be useful for Google calendars. Also, `sabre/dav` [implements Bearer tokens with version 3.1](http://sabre.io/dav/upgrade/3.0-to-3.1/). 

Do you want me to add this as well?

MichMich, 2017-03-06:
If it isn't a large refactor of the whole system (an no breaking api changes), please do.  Thanks!

deg0nz, 2017-03-07:
The pull request for this issue is #754

MichMich, 2017-03-17:
Merged!

closed 2017-03-17 by MichMich