MagicMirrorOrg/MagicMirror #752: Missing translation for weather forecast

BerndKohl, 2017-03-06:
When setting language to 'de' in global config.js the table header of weather forecast module still says 'WEATHER FORECAST'

Should be something like 'Wettervorhersage' in case 'appendLocationNameToHeader' is not set and 'Wettervorhersage für' if 'appendLocationNameToHeader' is set to true.

fewieden, 2017-03-07:
there is no header set, you have to do it yourself

```
{
  module: "weatherforecast",
  position: "top_right",
  header: "Wettervorhersage",
  config: {
    //your config
  }
}
```

in case you set ```appendLocationNameToHeader``` the location will be appended to your string you defined in the header

BerndKohl, 2017-03-07:
Thank you, missed that somehow.

closed 2017-03-07 by BerndKohl