MagicMirrorOrg/MagicMirror #753: Possibility to add '°C' to temperature values

BerndKohl, 2017-03-06:
It would be very much appreciated if there was a setting allowing the configuration of the displayed temperature values in modules weather forecast and current weather.

Currently only ° is display (in current weather) or nothing at all (in forecast).

It would like to have °C displayed after the temperature. e.g. 8°C or 45°F.

Speedy-one, 2017-03-09:
Just open the currentweather.js and search for 

`temperature.innerHTML =" " + thies.temperature + "&deg;";`

and add an **C** after &deg;

`temperature.innerHTML =" " + thies.temperature + "&deg;C";`

BerndKohl, 2017-03-09:
Thanks you.
But wouldn't that be overwritten when this module gets an update?
I would prefer a configurable solution...

amcolash, 2017-03-13:
See #773. Once pulled it should be in develop branch. Otherwise you can use the patch until merged.

BerndKohl, 2017-03-13:
Thanks a lot, that sounds very promising.

closed 2017-03-13 by BerndKohl

amcolash, 2017-03-13:
Glad to help!