MagicMirrorOrg/MagicMirror #771: Electron js/electron.js fails after firmware upgrade (Jessie Feb 20 release)

jdallasschwartz, 2017-03-12:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 3 model B, Raspbian GNU/Linux 8.0 (jessie), Linux magicpi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

**Node Version**: magicmirror@2.1.0 /home/pi/MagicMirror
└─┬ check-node-version@2.0.1
  ├── map-values@1.0.1
  ├── object-filter@1.0.2
  └── run-parallel@1.1.6

**MagicMirror Version**: 2.1.0

**Description**: Ran a apt-get update/upgrade around March 1st 2017 and after the update magic mirror would not work and gave the following message:

>"npm start
>
> magicmirror@2.1.0 start /home/pi/MagicMirror
> electron js/electron.js
>
>/home/pi/MagicMirror/node_modules/electron/index.js:9
> throw new Error('Electron failed to install correctly, please delete node_modules/' + >path.basename(__dirname) + ' and try installing again')
>  ^
>
>Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
>    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:9:9)
>    at Module._compile (module.js:570:32)
>    at Object.Module._extensions..js (module.js:579:10)
>    at Module.load (module.js:487:32)
>    at tryModuleLoad (module.js:446:12)
>    at Function.Module._load (module.js:438:3)
>    at Module.require (module.js:497:17)
>    at require (internal/module.js:20:19)
>    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:16)
>    at Module._compile (module.js:570:32)
>
>npm ERR! Linux 4.4.50-v7+
>npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
>npm ERR! node v6.10.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.1.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
>
>npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log"

**Steps to Reproduce**: Upgrade Raspbian Jessie to February 20th release, reboot pi, navigate to MagicMirror directory, run "npm start"

**Expected Results**: Magic Mirror screen (all modules worked before I ran the apt-get update/upgrade)

**Actual Results**: Error message described above

**Configuration**: " Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
        port: 8080,
        ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1", "x.x.x.x" ],

        language: 'en',
        timeFormat: 12,
        units: 'metric',

        modules: [
                {
                        module: 'alert',
                },
                {
                        module: "updatenotification",
                        position: "top_bar"
                },
                {
                        module: 'clock',
                        position: 'top_left'
                },
                {
                        module: 'calendar',
                        header: 'US Holidays',
                        position: 'top_left',
                        config: {
                                calendars: [
                                        {
                                                symbol: 'calendar-check-o ',
                                                url: 'REMOVED'
                                        }
                                ]
                        }
                },
                {
                        module: 'compliments',
                        position: 'lower_third'
                },
                {
                        module: 'currentweather',
                        position: 'top_right',
                        config: {
                                location: 'REMOVED',
                                locationID: 'REMOVED',  //ID from http://www.openweathermap.org
                                appid: 'REMOVED'
                        }
                },
                {
                        module: 'weatherforecast',
                        position: 'top_right',
                        header: 'Weather Forecast',
                        config: {
                                location: 'REMOVED',
                                locationID: 'REMOVED',  //ID from http://www.openweathermap.org
                                appid: 'REMOVED'
                        }
                },
                {
                        module: 'newsfeed',
                        position: 'bottom_bar',
                        config: {
                                feeds: [
                                        {
                                                title: "REMOVED",
                                                url: "REMOVED"
                                        }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },
                {
                        module: 'helloworld',
                        position: 'top_bar',
                        config: {
                                text: 'REMOVED',
                        }
                },
                {
                        module: 'MMM-Traffic',
                        position: 'bottom_right',
                        config: {
                                api_key: 'REMOVED',
                                mode: 'driving',
                                origin: 'REMOVED',
                                destination: 'REMOVED',
                        }
                },

        ]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
"

**Additional Notes**: Do I just need to go back to the release before the upgrade? I am relatively new to Linux so if you could provide the command to revert back that would be most helpful. Thank you very much for creating this and for your help!

roramirez, 2017-03-12:
This ISO included Pixel?

Please read the follow link https://github.com/MichMich/MagicMirror/issues/577#issuecomment-275478334

jdallasschwartz, 2017-03-13:
Hi roramirez. Yes, it appears that I am now running Pixel. I bought the Pi last year and updated it in early February 2017 before I started my Magic Mirror project. I had it all working and, like a dummy, decided to run another update a couple weeks ago which produced the error message above.

Thank you for posting that link. For some reason I couldn't find that before I posted this.

MichMich, 2017-03-17:
Can I close this issue, or is it still relevant?

roramirez, 2017-03-17:
I have not tested the Raspbian Febrary 2017. I think the issue can be try use without Pixel or other X window system.

@jdallasschwartz Should be add more information about that.

Many problems like this is related I mentioned before. https://github.com/MichMich/MagicMirror/issues/577#issuecomment-275478334

@MichMich  You can close.  If @jdallasschwartz can provide more information in the future  can reopen or open a new issue.

closed 2017-03-17 by MichMich