MagicMirrorOrg/MagicMirror #772: Calendar issues with All Day events

amcolash, 2017-03-13:
I have been having some issues with the calendar module showing incorrect dates for all day events. Below is an example for an event that should start at 12am (CST) on March 13, but it instead is showing as happening today. I am assuming it is something to do with the iCal parsing that is happening. I think I actually patched this in an older version of my own repo I had forked but have lost the patch. Has anyone else noticed this odd behavior?

BEGIN:VEVENT
DTSTART;VALUE=DATE:20170313
DTEND;VALUE=DATE:20170314
DTSTAMP:20170312T235617Z
UID:003uljj8ahler49h9bq6n61et0@google.com
CREATED:20170307T174910Z
DESCRIPTION:
LAST-MODIFIED:20170310T210214Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:All Day Event
TRANSP:TRANSPARENT
END:VEVENT

Also, I am using the latest develop branch and google calendar all day events.

amcolash, 2017-03-13:
Looking at the dates, they seem to be in order 3/13/2017 to 3/14/2017. Not entirely sure how the time stamp works

amcolash, 2017-03-13:
With some investigation, I am finding the event object made. What is interesting from this is that the start and end are separated and also have a timezone, where it appears the original didn't have the -5 timezone (cst).

```
{
    "type": "VEVENT",
    "params": [],
    "start": "2017-03-13T05:00:00.000Z",
    "end": "2017-03-14T05:00:00.000Z",
    "dtstamp": "20170313T015350Z",
    "uid": "003uljj8ahler49h9bq6n61et0@google.com",
    "created": "20170307T174910Z",
    "description": "",
    "last-modified": "20170310T210214Z",
    "location": "",
    "sequence": "0",
    "status": "CONFIRMED",
    "summary": "All Day Event",
    "transparency": "TRANSPARENT"
}
```

amcolash, 2017-03-13:
Logging the actual timestamp for the event gives me: `1489381200000`.

Passing this into [Epoch Converter](https://www.epochconverter.com/) gives me:
```
Assuming that this timestamp is in milliseconds:
GMT: Mon, 13 Mar 2017 05:00:00 GMT
Your time zone: 3/13/2017, 12:00:00 AM GMT-5:00 DST
```

which does not make sense, as it says that the event is today.

"today" looks to be defined as `1489298400000`
```
Assuming that this timestamp is in milliseconds:
GMT: Sun, 12 Mar 2017 06:00:00 GMT
Your time zone: 3/12/2017, 12:00:00 AM GMT-6:00
```

I am stumped. I wonder if it is because DST changed today (sigh, I wish there was not DST).

amcolash, 2017-03-13:
Well, thinking about all of this I understand now what is going on. When we ask for the "start" of the day, it is 0:00 that day. We then add 24 hours to make it 1 day of time. The issue is then with DST itself, where at 2:00am an hour is removed (or added). Therefore, I was effectively both time traveling :wink:  and also changing time zones.

I don't think that you can specify a timezone on an all day event, so this bug applies to only 2 days of the year. It still is worth noting that it is an issue.

In a slightly off-track comment, I noticed that we include momentJS, but then rarely actually use it and are doing date calculations by hand. We might have actually noticed if this was an issue if we did something like `moment().isBetween('2017-3-13', '2017-3-14');`. I am assuming there is more overhead with using moment instead of simple math calculations, but it might help in cases like this.

At least everything looks correct on my calendar today :smile:

MichMich, 2017-09-06:
Closed due to inactivity. Feel free to reopen.

closed 2017-09-06 by MichMich