MagicMirrorOrg/MagicMirror #774: Clarification of PM2 startup script for Jessie Lite

xbmcnut, 2017-03-13:
**Platform**:  Raspberry Pi2

**Node Version**: v6.10.0

**MagicMirror Version**: V2

**Description**: The instructions for Jessie Lite indicate the code required to input after running `pm2 startup` is `sudo su -c "env PATH=$PATH:/usw/bin pm2 startup linux -u pi --hp /home/pi"` but when I run `pm2 startup` on my instance, it asks me to enter `sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi`.

Could you kindly clarify which one is correct as auto start is not working for me?

xbmcnut, 2017-03-15:
I have now tried a full rebuild with a clean install of the full Jessie build now then leaned it out using [this guide](https://forum.magicmirror.builders/topic/236/complete-setup-tutorial). As before, once I start going through with the PM2 stuff, everything seems to work OK until a reboot and MM does not start. Surely someone else has seen this? Am I making a simple NOOB mistake here or does it just not work?

`pm2 list` returns: 
```
pi@raspberrypi:~ $ pm2 list
[PM2] Spawning PM2 daemon with pm2_home=/home/pi/.pm2
[PM2] PM2 Successfully daemonized
┌──────────┬────┬──────┬─────┬────────┬─────────┬────────┬─────┬─────┬──────────┐
│ App name │ id │ mode │ pid │ status │ restart │ uptime │ cpu │ mem │ watching │
└──────────┴────┴──────┴─────┴────────┴─────────┴────────┴─────┴─────┴──────────┘
 Use `pm2 show <id|name>` to get more details about an app
```

BerndKohl, 2017-03-15:
I can confirm this running MM2.1 on Jessie (NOT Lite).
Setting up MM according to the documentation does NOT restart MM after a reboot.

xbmcnut, 2017-03-15:
@BerndKohl Thank you for the confirmation. Glad to know I'm not going mad.

jnphilipp, 2017-03-15:
I had the same problem. I ended up adding the line `pm2 start /home/pi/mm.sh` to `~/.config/lxsession/LXDE-pi/autostart`. Autostart works fine now.

BerndKohl, 2017-03-15:
Seems to be related to: https://github.com/Unitech/pm2/issues/2765

roramirez, 2017-03-15:
Well, Wiki page have a wrong sentence

It said the next sentence:
`sudo su -c "env PATH=$PATH:/usw/bin pm2 startup linux -u pi --hp /home/pi"`

The correct sentence will be /usr/bin path not /usw/bin
`sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"`

If somebody can test this and fix the wikipage will be good.

xbmcnut, 2017-03-15:
@f-hj nailed this on the PM2 repository. 
#https://github.com/Unitech/pm2/issues/2765#issuecomment-286778941

Changed pm2-root.service timeout from 8 to 15s at `/etc/systemd/system/pm2-root.service` and that resolved my issue on Pi2.

MichMich, 2017-03-17:
Can I close this issue?

roramirez, 2017-03-17:
I'm fix the wrong path in wiki page
https://github.com/MichMich/MagicMirror/wiki/Jessie-Lite-Installation-Guide

if @xbmcnut can add a note how He fixed your problem in wiki page I think we can close this.

BerndKohl, 2017-03-18:
As far as I can tell the file is named
`/etc/systemd/system/pm2-pi.service`
not 
`/etc/systemd/system/pm2-root.service`

xbmcnut, 2017-03-19:
@roramirez Done.

roramirez, 2017-03-19:
Thanks @xbmcnut !

@MichMich you can close this issue.

closed 2017-03-31 by MichMich