MagicMirrorOrg/MagicMirror #776: Calendar 24 hours

olikami, 2017-03-14:
Infos

**Platform**: Raspberry Pi 3

**Node Version**: v6.10.0

**MagicMirror Version**: v2

**Description**: The calendar shows times as am/pm instead even if timeformat is set to 24.

**Steps to Reproduce**: Display any calendar with the calendar module.

roramirez, 2017-03-15:
I can see, this feature is not include into calendar module. I mean, the variable set `timeFormat` in global configuration variable isn't have effect the format how the  times are showing.

The config for `timeFormat` of **calendar module** have totally different behavior than global variable.

It's not difficult implement the behavior expected. I think this issue can be open until a contributor (or new) can add this.  

Add label as "first contribution" or "easy commit" can be an idea.

Richard238, 2017-03-25:
Same issue here. Added event to Google calendar for 26/03/2017 at 22:35
MM calendar shows it, incorrectly, as 10:35PM
MM `timeFormat` is set to 24
MM clock shows 24hr format as expected

roramirez, 2017-03-25:
HI @Richard238 
This is something we knowed. You can read in my previous comment.
The issue is open at someone can resolve it. If you or someone else are interested to fix go ahead with a Pull Request.

amcolash, 2017-03-26:
There was a previous issue on this #412. It looks like the easiest way to do this is by changing your language configuration to `en-gb`. I am not sure if there is a need to actually build this functionality out.

Just for those interested in git and version control, check out https://git-scm.com/book/en/v1/Getting-Started
A pull request is basically a request to the repository owners to include new features/functionality/fixes that are not currently in the codebase.
https://yangsu.github.io/pull-request-tutorial/

amcolash, 2017-03-26:
This is a hackish way of working things out. It won't work for languages other than english (not sure if there are a lot of non-english users). Moment has these formats associated with languages, so it might be a bit tricky to have it work internationally.

`calendar.js`

```
// Set locale.
if (config.language === 'en' && config.timeFormat == 24) {
  config.language = 'en-gb';
}
moment.locale(config.language);
```

Richard238, 2017-03-29:
@amcolash That works perfectly, thank you! :-)

johanhammar, 2017-03-30:
The problem is that moment.js defines the longDateFormat as "LT : 'h:mm A'" for en (United States, the default locale), which is what is used by the calendar()-function. 

There are different approaches on how to fix the problem:
1. Don't use moment calendar()-function, instead implement the functionality in the Calendar Module.
2. Use something like

        moment.locale(config.language);

		if (config.timeFormat == "24" && config.language == "en") {
			moment.updateLocale(config.language, {
  				longDateFormat : {
					LT: "hh:mm"
				}
			});
		}

But then you have to provide a list with each language that you want to override the long date format for. 

3. It's also possible to overwrite the calendar format but it has the same effects as above. 
4. Is there a better solution? Maybe it's good enough to start with the english locale and add more if issues are found? 

See https://github.com/moment/moment/tree/497f918515ae6ab900f008f19523b1e4ae5e2450/src/locale for a complete list of supported locales for moment.js and their formats.

amcolash, 2017-03-31:
I think we could safely assuming that all configs that use 24 hour format could just use the following. I am not sure if there are other ways of expressing time in other locales?

```
 moment.locale(config.language);

 if (config.timeFormat == "24") {
 	moment.updateLocale(config.language, {
 		longDateFormat : {
 			LT: "hh:mm"
 		}
 	});
 }
```

johanhammar, 2017-03-31:
I think that is good enough. It should also handle the opposite, i.e. using a locale that defaults to 24 hour format, i.e. sv or en-gb, and setting the timeformat configuration to 12.

amcolash, 2017-04-02:
Here is an example handling many different cases 12 hour / 24 hour and also handling the reverse in cases where a normally 24 hour location would want 12 hour format. I am almost thinking that having a custom time formatting that is user defined might be more useful than these edge cases.

```
moment.locale(config.language);

var timeFormat;

switch(config.language) {
case "en":
case "sv":
  if (config.timeFormat == "24") {
    timeFormat= "hh:mm";
  }
  break;
case "en-gb":
  if (config.timeFormat == "12") {
    timeFormat = "h:mm A";
  }
default:
  break;
}

if (timeFormat) {
  moment.updateLocale(config.language, {
    longDateFormat : {
      LT: timeFormat
    }
  });
}
```

QNimbus, 2017-06-26:
This issue has been open quite some time; To me it seems that the `config.timeFormat` config option if given, should be respected over the `config.language` option since it's a specific wish from the user to display the time in a 12 or 24 hour format.

I have created a pull request which I think implements this accordingly.

QNimbus, 2017-07-02:
@olikami This was fixed in commit [3404ebb](https://github.com/MichMich/MagicMirror/commit/3404ebbbb8bca36d9c2c2cc465345389c1700e1b). If you can confirm that it now works as expected you can close this issue.

olikami, 2017-07-03:
Okay, did a quick and dirty test, seems to work.

closed 2017-07-03 by olikami