MagicMirrorOrg/MagicMirror #791: Errors with dates before Jan 1, 1970

amcolash, 2017-03-25:
I think that I have some calendar events that somehow start before Jan 1, 1970 - can't say I know which of them is to blame here but I am assuming that this is a birthday. I was looking through the logs and kept seeing:

```
0|mm       | Error: dates lower than Thu Jan 01 1970 00:00:00 GMT-0600 (CST) are not supported
0|mm       |     at Object.toOrdinal (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:105:15)
0|mm       |     at Iterinfo.rebuild (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:1409:35)
0|mm       |     at RRule._iter (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:978:10)
0|mm       |     at RRule.between (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:794:21)
0|mm       |     at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:112:24
0|mm       |     at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:5)
0|mm       |     at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:188:22)
0|mm       |     at emitTwo (events.js:106:13)
0|mm       |     at Request.emit (events.js:191:7)
0|mm       |     at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1171:10)
```

I think that we should probably gracefully change their start dates to Jan 1, 1970 (if we are not calculating ages when showing B-Days). I might have some time in the next few days to work on a PR.

roramirez, 2017-03-25:
Here is a thing related.
https://github.com/jakubroztocil/rrule/pull/125

I don't if the rrule project  include this into and what kind impact can will be. Are you propose keep a rrule patched for this kind of case?

amcolash, 2017-03-26:
Interesting. I am not sure of the impact of changing dates we try and parse. If that causes issues I think that it would be best to have a patched rrule version if that PR is not merged.

amcolash, 2017-03-26:
Doing some more research into the issue, looks like JS is more than capable of dealing with dates before 1970 (since the min/max dates able to be defined are much larger than that). http://stackoverflow.com/questions/27093130/how-to-get-the-minimum-and-maximum-date

amcolash, 2017-03-26:
I edited my local version of rrule by changing `dateUtil.ORDINAL_BASE = new Date(1970, 0, 1);` to `dateUtil.ORDINAL_BASE = new Date(1900, 0, 1);` which works for me. I am assuming that there is a better way. Without this hack however, I am not able to see some of my calendar events because rrule kills the calendar thread early.

roramirez, 2017-04-15:
This is related with  #839. Some calendar are not loaded maybe containt events before Jan 1, 1970

tgeimer, 2017-04-18:
Thank you! changing the ORDINAL_BASE worked for me.

nhubbard, 2017-04-22:
This issue is called [epoch time](https://en.wikipedia.org/wiki/Unix_time). It is how Unix (and by extent, Linux) systems count time. It starts on January 1<sup>st</sup>, 1970. This issues makes complete sense, considering that negative time doesn't exist.

berlincount, 2017-04-22:
Sorry, but I don't think limited representability in one of the underlying formats is a good reason to not support times outside that formats range ;)

roramirez, 2017-04-22:
@berlincount 
Recently I added  comment here about that
https://github.com/MichMich/MagicMirror/issues/839#issuecomment-296404724

nhubbard, 2017-04-22:
@berlincount I mean, yes, JavaScript supports going before that, but it doesn't make sense unless you patch the kernel. Which is annoying and weird. In this case, you *can* edit the ORDINAL_BASE as said above.

amcolash, 2017-04-23:
Seems like ruule-alt isn't much more supported than just rrule. I wonder if there are other repeating rule js libraries out there (haven't checked but might be worth it).

MichMich, 2017-09-06:
Closed due to inactivity. Feel free to reopen.

closed 2017-09-06 by MichMich