MagicMirrorOrg/MagicMirror #797: Unable to select timezone

xy86, 2017-03-26:
Not able to choose timezone. 

Tried:
- timezone: Etc/GMT+01
- timezone: Etc/UTC+02

....for Oslo - Norway.

The clock disappears whenever a timezone-value is added (besides "null" or missing parameter).

Any help appreciated :)

MichMich, 2017-04-05:
Merged. Closing this issue.

roramirez, 2017-04-06:
This issue I think need re-open, there a little thing I'm under investigation about mentioned in 
review https://github.com/MichMich/MagicMirror/pull/803#pullrequestreview-30370685

MichMich, 2017-04-06:
Reopened.

MichMich, 2017-09-06:
@roramirez Can I close this one?

MichMich, 2017-09-29:
Closed due to inactivity.

closed 2017-09-29 by MichMich

sushant96702, 2020-06-14:
Hi i have installed magic mirror recently and i wanna know that how can i change timezone i am from india 
please give me step by step instruction
any help would be appreciated

sdetweil, 2020-06-14:
the clock settings are in the base OS, not in MagicMirror

so, change your system time

sushant96702, 2020-06-16:
thanks but how do i change time format to 12 hours as on rasp pi i have 12 hr time format but time on magic mirror is still 24 hrs

sdetweil, 2020-06-16:
@sushant96702  edit the ~/MagicMirror/config/config.js and change the timeFormat to 12 from 24

```
        language: "en",
        timeFormat: 24,                   // < -- this
        units: "metric",
```

sushant96702, 2020-06-16:
i have already done this

sdetweil, 2020-06-16:
please post the config.js. note that the config is case sensitive.. timeformat is NOT the same as timeFormat

it should read
```
timeFormat: 12,
```
works for me

sushant96702, 2020-06-16:
[config.txt](https://github.com/MichMich/MagicMirror/files/4786702/config.txt)

i have turned into txt because git doesnt support .js files

sdetweil, 2020-06-16:
@sushant96702  thanks

I put this in for my config.js, by using cp from my Downloads (no edits)
and got this result  (can't post pictures here), working as expected
https://www.dropbox.com/s/3efaky0vf4q5k3x/Screenshot%20at%202020-06-16%2008-03-40.png?dl=0

sushant96702, 2020-06-16:
but why it aint working on my rasp pi
IDK why it works on yours but not on mine
please help

sdetweil, 2020-06-16:
tell me more how you have implemented MagicMirror? docker? how install? 

is this the actual config.js being used?  if u take out some quote mark (") does MM fail to load?

sushant96702, 2020-06-16:
i want to gift  it to my sister
i have installed it using the link which used to be on magic mirror website
yes it is actual config file
Which quotes  "if u take out some quote mark (") does MM fail to load"

sdetweil, 2020-06-16:
what version is this? do
```
grep version ~/MagicMirror/package.json
```

can u post a link to a  screenshot

sushant96702, 2020-06-16:
"version": "2.11.0",

sdetweil, 2020-06-16:
can u post a link to a screenshot of the MM page

sdetweil, 2020-06-16:
@sushant96702 

can u also try something
edit 
~/MagicMirror/modules/default/clock/clock.js
and change this line 
```
		timeFormat: config.timeFormat,
```
to
```
		timeFormat: 24,
```
then restart MM.. please advise

sushant96702, 2020-06-16:
https://www.dropbox.com/sh/iyn6yp5z2y447mu/AAC7u46pSnzLFlM-8vFHH13ya?dl=0

sdetweil, 2020-06-16:
thx.. weird, the weather shows a different base time (19:26).. than clock(19:01)

sushant96702, 2020-06-16:
Thanks i didnt noticed that BTW do you know why is it happening

sdetweil, 2020-06-16:
no.. never seen anything like this

what time IS it on the system?

sushant96702, 2020-06-16:
19.19 on rasp and on left side of MM
7:26 pm on weather wait what how the heck is it in 12 hr format
what is happening with my software

sdetweil, 2020-06-16:
that time on weather is sunrise or sunset time..  so different.. my image was also different..  
each module does their own thing.. not coordinated with base modules (system clock is everything)

still.. can u change that one line of code I suggested

sushant96702, 2020-06-16:
yep i did that but nothing changed

sdetweil, 2020-06-16:
change it to 12 and retest

sushant96702, 2020-06-16:
yes i changed it to 12 and rebooted it but it was still in 24 hr format

sdetweil, 2020-06-16:
how is MM started? pm2?  you didn't need to reboot, just pm2 restart ?? name/number
do
```
pm2 status
```

sushant96702, 2020-06-16:
has this thing ever occured to anyone

sushant96702, 2020-06-16:
yes i have put it into autorun

sdetweil, 2020-06-16:
in the 3 years I have been on MM I have never seen it from anyone

so I am wondering if u have 2 MM running.. 1st with some other config and this one starting second and failing cause the port is in use...

sushant96702, 2020-06-16:
sorry i didnt understood what you just said

sdetweil, 2020-06-16:
i have seen where there are two instances of MM installed, and both are marked to start at boot. 
both using the same port

the 1st one starts and displays
the 2nd one tries to start , but the port is in use (by the 1st), so it fails

I see this on my dev system when I forget that pm2 starts 1 at boot and I try to start another for testing something

sushant96702, 2020-06-16:
ID      name                  mode        @       Status       CPU       Memory
0        MagicMirror        Fork          0        Online       0%         2.4mb

sdetweil, 2020-06-16:
we should take this to the forum or discord better suited to chat format

sushant96702, 2020-06-16:
Sorry Sir but i dont know how to use forms

sushant96702, 2020-06-16:
And there is only one MM installed

sdetweil, 2020-06-16:
>yes i have put it into autorun

what does that mean? 

u posted the pm2 status output

sushant96702, 2020-06-16:
it asked in terminal that "do you want autorun(y/n)'

sdetweil, 2020-06-16:
hm.. i maintain the installer script, and it doesn't have that message
mine has 
"Do you want use pm2 for auto starting of your MagicMirror (y/N)?"

sushant96702, 2020-06-16:
sorry it said same as yours i forgot

sushant96702, 2020-06-16:
please help and tell me a solution

sdetweil, 2020-06-16:
one more thing can u show me the output of
```
ls ~/MagicMirror/config -laF
```

sdetweil, 2020-06-16:
@sushant96702   waiting for info

sushant96702, 2020-06-16:
@sdetweil i am soo sorry sir i was not able to end you info can i just send it to you tommo because it is 1:00am