MagicMirrorOrg/MagicMirror #807: fulldayevents from outlook.com calendar

yawnsde, 2017-03-31:
I'm not quite sure if I should mention this as an issue or submit a pull request. I use an outlook.com calendar and noticed that fullDayEvents are not marked as fulldayevent by calendarfetcher.js

This is because of the `isFullDayEvent` function.

Example: I have a fullDayEvent on my calendar for holiday, starting on April 1st ending on April 10th, marked as a fullDayEvent. isFullDayEvent now checks this entry and neither `event.start.length === 8` nor the calculation for 24h is valid. I added a new if clause so my function now looks like this:

```javascript
	/* isFullDayEvent(event)
	 * Checks if an event is a fullday event.
	 *
	 * argument event obejct - The event object to check.
	 *
	 * return bool - The event is a fullday event.
	 */
	var isFullDayEvent = function(event) {
		if (event.start.length === 8) {
			return true;
		}
		
		if (event['MICROSOFT-CDO-ALLDAYEVENT'] === 'TRUE') {
			return true;
		}

		var start = event.start || 0;
		var startDate = new Date(start);
		var end = event.end || 0;

		if (end - start === 24 * 60 * 60 * 1000 && startDate.getHours() === 0 && startDate.getMinutes() === 0) {
			// Is 24 hours, and starts on the middle of the night.
			return true;
		}

		return false;
	};
```

The outlook.com calendar engine seems to add the attribute `'MICROSOFT-CDO-ALLDAYEVENT'` with single quotation marks and it is not set to boolean (true or false) but to `'TRUE'` or `'FALSE'`. This works fine for me and did not cause any trouble with other calendars. Maybe you want to add this to the repository?

MichMich, 2017-04-01:
Yes, please send a PR (including updated Changelog)

MichMich, 2017-09-06:
Closed due to inactivity. Feel free to reopen.

closed 2017-09-06 by MichMich