MagicMirrorOrg/MagicMirror #819: [RFE] provide possibility to relocate a module node

McFoggy, 2017-04-03:
**Platform**: not relevant, working on windows targetting RaspberryPi

**Node Version**: not relevant, working on v6.10.0

**MagicMirror Version**: v2, forked

**Description**: 

as a _module designer_ I would like that, at runtime , my module has the possibility to relocate itself in a new predefined location ; example from `bottom_center` to `fullscreen_above`.

I would probably require an API like the following:

```
	/* relocateModule(module, target, hideSpeed, retargetCallback, showSpeed, finalCallback)
	 * Relocate the module in another zone.
	 *
	 * argument module Module - The module to relocate.
	 * argument target String - location of the target zone (one of "top_bar", "top_left", "top_center", "top_right", "upper_third", "middle_center", "lower_third", "bottom_left", "bottom_center", "bottom_right", "bottom_bar", "fullscreen_above", "fullscreen_below")
	 * argument hideSpeed Number - The speed of the hide animation (in milliseconds).
	 * argument relocateCallback function - Called when the module DOM wrapper has been re-attached to its new location
	 * argument showSpeed Number - The speed of the show animation (in milliseconds).
	 * argument finalCallback function - Called when the new element is shown in its nw place
	 */
	function relocateModule(module, target, hideSpeed, relocateCallback, showSpeed, finalCallback)
```

**Expected Results**: Having this in place I could have one module being able to change its UI & UI location depending on some context.

**Actual Results**: To simulate the same currently, one has to use several modules (one for each place) and coordinate hide/show of all of them.

MichMich, 2017-04-03:
The position is currently up to the user. Can you give any use cases in which this is interesting?

McFoggy, 2017-04-04:
> The position is currently up to the user.

the position is up to the integrator of the system, for me the user is the one in front of the mirror using it.
Depending on what you use the library for this makes a big difference. I am working in healthcare for a software company, we want to use your framework to deliver new functionnalities to our end users (physicians, nurse, surgeon, hospital members), so we do the integration and the users are 'just' using the mirror as it is provided.

> Can you give any use cases in which this is interesting?

of course, currently I am building a system with vocal regonition where modules can on the fly register/unregister vocal commands depending on their context/workflow. This vocal recognition is also a module that, by default, shows a message at the bottom of the screen. When the user says 'help me', the list of registered commands is shown full screen.
As I said the feature can be easily by passed by using several 'modules' and playing with the visibiity of each but then you have related functionnalities/domain object/.... that are splitted in different modules which is not very cool for maintenance.

fewieden, 2017-04-04:
@McFoggy I made voice recognition modules for MM where I do some similar action.

On the command OPEN HELP there is a list with the voice commands for this module on top of all other modules and the background (all modules) get blurred behind it

McFoggy, 2017-04-04:
@fewieden thanks , I looked at your module and indeed it is a lot easier to use a fixed additonal DOM element than relocating a changing module DOM element.

McFoggy, 2017-04-04:
@MichMich I'll use @fewieden technic instead, it is a lot easier than relocating, forget the RFE.

closed 2017-04-04 by McFoggy

McFoggy, 2017-04-05:
@MichMich For the help scenario I used @fewieden technic, which worked perfectly.

But now I have new business scenarios in which using above technic is a total hack.

My setup is the following (simplified):
- a non visual module for voice recognition
- a module A for a particular business case  (that is a workflow with long duration), shown normally centrally in mirror with icons & texts to explain the workflow steps.
- X modules for detailed information that also need to render centrally

While the workflow progress, via vocal commands the user can switch/activate other modules. Here I'd like to move the workflow from the central place to the bottom left (for example) in a reduced mode/reduced size.

You could argue that the detailed module could come above the module A and be hidden afterwards, but once started, I cannot dismiss anymore the module A.

We can imagine that module A is a 'football match video' and meanwhile you watch it, you want to look at your emails but you do not want to miss any goal ; the video will thus go in thumbnail format in a small part of the screen.

Long story to make it short, I think that in that case I need a relocation feature. What do you think? Should I reopen the issue?

I'll try to implement the feature in a fork an will propose it as a PR.

MichMich, 2017-04-05:
I think it's an extreme edge-case which you can solve using other techniques. I personally don't think it's worth an API change.

McFoggy, 2017-04-06:
>  I personally don't think it's worth an API change.

No problem, I understand.

> ... you can solve using other techniques

do you have things in mind? I probably missed easier techniques.

fewieden, 2017-04-06:
you could do this with css classes
class video -> transition on position and size
then add the thumbnail class
class thumbnail -> new position, new size

McFoggy, 2017-04-06:
changing the dom css classes would not be enough, would it?

Below the clock is inserted in the `top left` bar
[image]

If I want it `bottom right` at runtime, it is not enough to change the dom css classes. From what I understand of the MM layouting, regions are layed-out via `absolute` positionning and modules' DOM nodes are inserted inside those regions and should use a relative positioning.

I'll investigate further the layouting internals to better understand what you propose.