MagicMirrorOrg/MagicMirror #826: Module calendar: titleReplace not working anymore?

BerndKohl, 2017-04-05:
**Platform**: Raspberry Pi 3

**Node Version**: 6

**MagicMirror Version**: 2.1.1

**Description**: I am using titleReplace in calendar module to shorten some title (e.g. "Geburtstag" to "Geb."). This worked for some time but suddenly (cannot tell if it relates to the upgrade of MM from 2.1 to 2.1.1) it stopped working. Now no title gets replaced by its substitution - no changes done to config.js.

**Steps to Reproduce**: add titleReplace in config.js for module calendar and replace e.g. "Geburtstag" by "Geb.". Note that you will need some full-day events in your calendar (in my case: birthdays)

**Expected Results**: "Tina 32. Geburtstag" gets replaced with "Tina 32. Geb." in calendar view

**Actual Results**: No replacement. "Tina 32. Geburtstag" stays "Tina 32. Geburtstag"

MichMich, 2017-04-05:
I noticed the same on my calendar with the Facebook birthdays recently. I haven't yet upgraded to 2.1.1 though. Maybe it's a Facebook specific issue. I'd need to investigate this further.

If somebody discovers the issue, feel free to send a PR.

BerndKohl, 2017-04-05:
I am using a calendar in iCloud.
Correction: I am running NodeJS 6.10.2

MichMich, 2017-04-06:
I tried it on may side and it is working on the current develop branch. Could you share the configuration section of your calendar (minus the private details). Thanks!

BerndKohl, 2017-04-06:
module: "calendar",
      header: "Bevorstehende Ereignisse",
      position: "top_left",
      titleReplace: {   
          "Geburtstag": "Geb.",         
          "DTV Diespeck": "DTV"         
        },                      
      config: {         
        maximumEntries: 7,      
        colored: true,          
        dateFormat: "D.M.",     
        calendars: [            
          {                             
            symbol: "clock-o", //calender       
            color: "#ffff00",                   
            url: "https://p03-calendarws.icloud.com/ca/subscribe..."
          },
          {                             
            symbol: "soccer-ball-o", //soccer
            color: "#F2F5A9",                   
            url: "https://p24-calendars.icloud.com/published/..."
          },                                    
          {
            symbol: "calendar-check-o", // birthdays
            color: "#58ACFA",           
            url: "https://p24-calendars.icloud.com/published/..."
          }
        ]
      }

DutchDynamics, 2017-04-07:
I noticed that none of the setting on the calendar module is working anymore.
For example changing the fadePoint value, did not work. But also the config for displayRepeatingCountTitle and colored are not working. 
Only when changing the default value in calendar.js it starts working. The individual values on calendar level are working correct.

MichMich, 2017-04-07:
@BerndKohl Your titleReplace setting should be part of the config object.

MichMich, 2017-04-07:
@DutchDynamics: Please share your config. I'm 99% sure its a config error.

BerndKohl, 2017-04-07:
@MichMich Thank you, that did the trick! Up and running!

DutchDynamics, 2017-04-07:
module: "calendar",
			header: "What's Next",
			position: "top_right",
			displayRepeatingCountTitle: true,
			classes: "default everyone",
			getRelative: 48,
			fadePoint: 0.1,
			colored: true,
			config: {
				calendars: [
					{
						color: "#FFFFFF",
						symbol: "calendar-check-o ",
						url: "https://outlook.office365.com/owa/c..."
					},
					{
						color: "#FF8888",
						symbol: "car ",
						url: "https://www.f1calendar.com/d...."
					}
				]
			}

MichMich, 2017-04-07:
Same issue. All your config settings should be part of the config object.

DutchDynamics, 2017-04-07:
Dank je wel (Thanks)

closed 2017-04-07 by MichMich