MagicMirrorOrg/MagicMirror #831: [Module: Current Weather] Add support for daily Min & Max Temperature

winstonma, 2017-04-06:
In the OpenWeather API call there is "main.temp_max" and "main.temp_min". It would be great if Daily Min/Max is being displayed on Magic Mirror.

Thanks

MichMich, 2017-04-12:
I might be confused, but the weather forecast already shows these values?

winstonma, 2017-04-13:
I guess the weather forecast should show today's min max but it didn't.

BerndKohl, 2017-04-13:
You got me confused: @winstonma is relating to Current Weather module (according to the title of the thread) whereas @MichMich is talking about weather forecast.
If I get it right @winstonma wants to be able to show todays min and max in Current Weather, NOT in Weather Forecast (where it is already being shown.). This would allow showing todays min/max/current without enabling Weather Forecast module.

winstonma, 2017-04-13:
Thanks @BerndKohl you are right but I am not really mind as long as one of them shows the min/max.

But I can't see today's min/max in any of the modules so I created this issue.

winstonma, 2017-04-16:
I just checked this afternoon and it shows today's weather forecast. So I am not sure what happened.

Does TimeZone affect the result?

winstonma, 2017-05-18:
I don't know why but seems it is working now.

I think it is better to close this.

closed 2017-05-18 by winstonma