MagicMirrorOrg/MagicMirror #834: after update black Screen

Beginner88888, 2017-04-06:
hi guys,

today i had made first time an update. From the directory of MagicMirror with git pull. 
after them i had a black screen. 
When i start npm out of the MagicMirror Folder is got following issue record.

WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'rrule-alt'
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
App threw an error during load
Error: Cannot find module 'rrule-alt'
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:20:13)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
Whoops! There was an uncaught exception...
{ Error: Cannot find module 'rrule-alt'
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:20:13)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12) code: 'MODULE_NOT_FOUND' }

fewieden, 2017-04-06:
did you install the dependencies? ```npm install```

yomgdf, 2017-04-07:
Hello,

Same thing happen to me. 
I try to install dependencies as fewieden asks and its works.

Beginner88888, 2017-04-07:
Hi 

I switched into the folder magic mirror and execute
"Npm install" 

No changes.

fewieden, 2017-04-07:
can you double check if you entered the correct directory `cd ~/MagicMirror`, then `npm install` as it should install `"rrule-alt": "^2.2.3",` as listed in the package.json. If this still doesn't work I suggest to create a topic in the forum

Beginner88888, 2017-04-07:
Ok, after installing npm, i get now following mesage by starting "npm start" out of folder "Magic Mirror"

Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE :::8080
    at Object.exports._errnoException (util.js:1026:11)
    at exports._exceptionWithHostPort (util.js:1049:20)
    at Server._listen2 (net.js:1253:14)
    at listen (net.js:1289:10)
    at net.js:1399:9
    at _combinedTickCallback (internal/process/next_tick.js:77:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)
    at Module.runMain (module.js:592:11)
    at run (bootstrap_node.js:402:7)
    at startup (bootstrap_node.js:157:9)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080 }

Beginner88888, 2017-04-07:
i have found the error.
my wunderlist was at position left. 
after change to top_left everything was fine

closed 2017-04-07 by Beginner88888