MagicMirrorOrg/MagicMirror #835: Docker Setup

khassel, 2017-04-06:
Hi,

trying to setup docker in server only mode. Executing 

`docker run  -d --publish 80:8080 --restart always --volume ~/magic_mirror/config:/opt/magic_mirror/config --volume ~/magic_mirror/modules:/opt/magic_mirror/modules --name magic_mirror MichMich/MagicMirror
`

results in

`docker: Error parsing reference: "MichMich/MagicMirror" is not a valid repository/tag: repository name must be lowercase.
`

Next try with lowercase

`docker run  -d --publish 80:8080 --restart always --volume ~/magic_mirror/config:/opt/magic_mirror/config --volume ~/magic_mirror/modules:/opt/magic_mirror/modules --name magic_mirror michmich/magicmirror
`

results in

```
Unable to find image 'michmich/magicmirror:latest' locally
docker: Error response from daemon: repository michmich/magicmirror not found: does not exist or no pull access.

```

So where is the image to find in the docker store?

I did the build myself with

`docker build -t magic_mirror .
`

and now I can start the container.

Two Problems:

1. For accessing the container with "localhost:8080" as described for the manual setup you have to change the parameter `--publish 80:8080` to `--publish 8080:8080`

2. With every start of the container the mounted files outside of the container are overwritten with the files from inside the container. This is already described in [this pull request](https://github.com/MichMich/MagicMirror/pull/833) for the config-directory, same issue exists for the modules-directory.

Thanks,

Karsten.

bastilimbach, 2017-04-12:
@khassel Take a look at my PR (#846). I fixed the overwriting of the modules folder and changed the README. Now, @MichMich only needs to publish this image on Docker Hub/Store and it should all work. Hopefully 👍

roramirez, 2017-04-15:
Nice work @bastilimbach 

I've thinking about an idea. It wold be cool if we can integrate an arm image like Debian + Pixel and run the test suite there integrated directly in Travis or other CI. 

Are you have some idea how we can do it this?

bastilimbach, 2017-04-15:
Thanks @roramirez 😊 
Correct me if I'm wrong. So you want to run the testing within the docker container/build process?

roramirez, 2017-04-15:
Yes. the idea is to equals architecture and OS for "supported system", example Raspbian Jessie + Pixel.

bastilimbach, 2017-04-15:
Ok but I don't get the point, why this is needed. If the OS has Docker installed, the application will work. There is just one possible event, where the application doesn't work and that's when the CPU architecture is ARM or x86 and the Docker Image uses a non ARM or x86 base os. Or what do you mean with `"equals architecture"`?

roramirez, 2017-04-15:
I mean with "equals architecture"  for example, have a Docker Image with _Raspbian Jessie + Pixel_  and  run the test suite in this  defined container like as we are doing in Travis.

bastilimbach, 2017-04-15:
Ah, I think I get it. So we create different Dockerfiles which have all different kinds of OS, right? Then we run a little test suit and see if electron is starting correctly etc. And on every new release, we run these Docker Images to test different operating systems and we can have a little "supported operation systems" section in the readme.

I think we can accomplish this within travis ci. I need to check if I'm right but I think you can have "stages" which can be executed in different docker containers.

bastilimbach, 2017-04-15:
@roramirez Take a look at this: https://docs.travis-ci.com/user/multi-os/
You can choose between Linux and macOS and what they have installed. But the linux version is always Ubuntu, which isn't optimal. I don't see a way of changing the Linux distro in a travis build. Do you?

PR #846 (2017-04-12): Add the ability to use the docker image with a raspberry pi
## What does this PR do?
This PR replaces the base image from the Dockerfile and fixes some little thinks (see related issues).
Due to the raspberry pi architecture, the current Docker image doesn't work with any Raspberry PI, because it uses Debian with an x86 architecture. I replaced Debian with an ARM compatible Linux distro, so the image also works on a Raspberry PI.
## Related issues
#835 #833 
## Important
@MichMich You need to connect this repo to your Docker Hub/Store account, so we can actually use [this command](https://gith [... cut]
closed 2017-04-20 by khassel