MagicMirrorOrg/MagicMirror #838: error in npm start

athuldevin, 2017-04-09:
root@athuldevin:~/Desktop/magic mirror/MagicMirror-master# npm start

> magicmirror@2.1.1 start /root/Desktop/magic mirror/MagicMirror-master
> sh run-start.sh

run-start.sh: 4: run-start.sh: electron: not found

npm ERR! magicmirror@2.1.1 start: `sh run-start.sh`
npm ERR! Exit status 127
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.1 start script.
npm ERR! This is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.6.0-kali1-amd64
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! cwd /root/Desktop/magic mirror/MagicMirror-master
npm ERR! node -v v4.7.2
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/Desktop/magic mirror/MagicMirror-master/npm-debug.log
npm ERR! not ok code 0
root@athuldevin:~/Desktop/magic mirror/MagicMirror-master# 

i tried in both Linux and windows ,node js and npm installed correctly

fewieden, 2017-04-09:
did you run npm install?

athuldevin, 2017-04-09:
yes i did by opening terminal in that folder

BerndKohl, 2017-04-09:
Try a reboot, helped for me.

roramirez, 2017-04-09:
Node  v4.7.2? I'm sure if it working there.  Try update you node version and try again.

athuldevin, 2017-04-10:
i started from beginning all the outputs are given below

`
C:\Users\win-7\Downloads\magic mirror\MagicMirror-master>node -v
v6.9.5

C:\Users\win-7\Downloads\magic mirror\MagicMirror-master>npm -v
4.4.4

C:\Users\win-7\Downloads\magic mirror\MagicMirror-master>npm install
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g
npm WARN prefer global colorguard@1.2.0 should be installed with -g

> electron-chromedriver@1.6.0 install C:\Users\win-7\Downloads\magic mirror\MagicMirror-master\node_modules\electron-chromedriver
> node ./download-chromedriver.js

> spawn-sync@1.0.15 postinstall C:\Users\win-7\Downloads\magic mirror\MagicMirror-master\node_modules\spawn-sync
> node postinstall

> electron@1.6.2 postinstall C:\Users\win-7\Downloads\magic mirror\MagicMirror-master\node_modules\electron
> node install.js

C:\Users\win-7\Downloads\magic mirror\MagicMirror-master\node_modules\electron\install.js:46
  throw err
  ^

Error: read ECONNRESET
    at exports._errnoException (util.js:1022:11)
    at TLSWrap.onread (net.js:569:26)
magicmirror@2.1.1 C:\Users\win-7\Downloads\magic mirror\MagicMirror-master
+-- body-parser@1.17.1
| +-- bytes@2.4.0
| +-- content-type@1.0.2
| +-- debug@2.6.1
| | `-- ms@0.7.2
| +-- depd@1.1.0
| +-- http-errors@1.6.1
| | +-- depd@1.1.0  deduped
| | +-- inherits@2.0.3
| | +-- setprototypeof@1.0.3  deduped
| | `-- statuses@1.3.1  deduped
| +-- iconv-lite@0.4.15  deduped
| +-- on-finished@2.3.0
| | `-- ee-first@1.1.1
| +-- qs@6.4.0
| +-- raw-body@2.2.0
| | +-- bytes@2.4.0  deduped
| | +-- iconv-lite@0.4.15  deduped
| | `-- unpipe@1.0.0
| `-- type-is@1.6.15
|   +-- media-typer@0.3.0
|   `-- mime-types@2.1.15  deduped
+-- chai@3.5.0
| +-- assertion-error@1.0.2
| +-- deep-eql@0.1.3
| | `-- type-detect@0.1.1
| `-- type-detect@1.0.0
+-- chai-as-promised@6.0.0
| `-- check-error@1.0.2
+-- colors@1.1.2
+-- express@4.15.2
| +-- accepts@1.3.3
| | +-- mime-types@2.1.15  deduped
| | `-- negotiator@0.6.1
| +-- array-flatten@1.1.1
| +-- content-disposition@0.5.2
| +-- content-type@1.0.2  deduped
| +-- cookie@0.3.1
| +-- cookie-signature@1.0.6
[... 1118 lines cut ...]
> sh run-start.sh

'sh' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.1.1 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.1.1 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\win-7\AppData\Roaming\npm-cache\_logs\2017-04-10T14_37_40_185Z-debug.log

C:\Users\win-7\Downloads\magic mirror\MagicMirror-master>`

athuldevin, 2017-04-10:
is the problem due to eletron installation .i don't know well

fewieden, 2017-04-10:
'sh' is not recognized as an internal or external command,

you cant run sh scripts out of the box in windows

alexf4, 2017-04-11:
I'm getting the same error. Fresh install

athuldevin, 2017-04-12:
i installed sh globally using "npm install -g sh"  no use

superfino, 2017-04-26:
Suffered the same problem with a fresh MM installation on a Rpi III. After starting MM the default modules are obviously running but MM struggles with additional modules like MMM-MotionEye.
Are there any ideas or solutions for this ???

`npmnpm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.1.1 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.1 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2017-04-26T12_24_40_129Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.1.1 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.1 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2017-04-26T15_24_37_429Z-debug.log`

MichMich, 2017-05-26:
All mentioned issues are non MagicMirror specific. Therefor I close this issue. If I need to reopen it, feel free to let me know.

closed 2017-05-26 by MichMich

athuldevin, 2017-05-27:
then can you tell me where is the problem

Joey-1970, 2018-02-14:
Hello!
I'm new at the MagicMirror-Project - but have the same issue. I take a new Raspberry Pi 3b, a new Raspian Stretch Image and try hours and hours to start MagicMirror, but have allways the "npm code ELIFECYCLE"-problem. I found this issue often, but I don't know the right solution. What is it?

Joachim

MichMich, 2018-02-14:
Joachim, did you start your Raspberry in X Desktop or in CLI (Command Line Interface)?

Joey-1970, 2018-02-14:
Hello MichMich,

I try both of them. First as CLI, after I read somewhere that use the desktop-option it will work, I do this. But after one successful start - and temporary hapiness - the same as before...

Joachim

MichMich, 2018-02-14:
In that case I suggest you post a support question on http://forum.magicmirror.builders.

sunnyg9361, 2018-08-15:
Joachim, do you have any update please. I have the same problem. Thanks alot for your help MichMich and Joachim

Hachaki, 2018-12-13:
pi@raspberrypi:~/MagicMirror $ npm start
(node:1165) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.

> magicmirror@2.5.0 start /home/pi/MagicMirror
> sh run-start.sh

run-start.sh: 4: run-start.sh: electron: not found

npm ERR! magicmirror@2.5.0 start: `sh run-start.sh`
npm ERR! Exit status 127
npm ERR! 
npm ERR! Failed at the magicmirror@2.5.0 start script.
npm ERR! This is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0

does anybody know what to do ?

mrkus21, 2019-03-19:
Same Problem here, anyone can help, please ?!

vishwapinnawala, 2019-11-16:
guys try  DISPLAY=:1 npm start

roramirez, 2019-11-20:
If you get the `electron: not found` should install dependencies.

Try before
`npm install`