MagicMirrorOrg/MagicMirror #839: Some calendars won't load

fewieden, 2017-04-09:
Lately a lot of users complain about that some of their calendar files wont be loaded anymore in v2.1.1

https://forum.magicmirror.builders/topic/2221/after-upgrade-v2-1-1-multiple-calendars-not-displaying
https://forum.magicmirror.builders/topic/2208/calendar-shows-no-entries-after-mm-update

coderaaron, 2017-04-11:
So, here is what I'm getting when I do `npm start dev`

```Launching application.
Create new calendar fetcher for url: https://outlook.office365.com/owa/calendar/<REDACTED>@<REDACTED>/S-1-8-117873446-3978456252-3857600756-3375899348/reachcalendar.ics - Interval: 300000
Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
Whoops! There was an uncaught exception...
TypeError: curr.exdates[i].toISOString is not a function
    at Object.ical.objectHandlers.END (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:44:44)
    at Object.handleObject (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:267:41)
    at Object.parseICS (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:308:20)
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:24)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:188:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1171:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues```

I will do some more investigating when I have time but I thought I'd report it in case anyone with more experience knows off the top of their head what the problem is.

coderaaron, 2017-04-12:
After a little more digging, it looks like this is caused by recurring events that have more than one `exdate` (exception date)...I started debugging the ical.js and node-ical.js, but need to do some work so I can't go any further today

DutchDynamics, 2017-04-16:
Hi, I also have the same problem. It looks like there are different ways to deal with multiple EXDATA in the ics file.
Some calendars return it like this:
EXDATE;TZID=South Africa Standard Time:20170403T060000,20170410T060000,20170417T060000,20170501T060000

others return it like:
EXDATE;TZID=South Africa Standard Time:20170403T060000
EXDATE;TZID=South Africa Standard Time:20170410T060000
EXDATE;TZID=South Africa Standard Time:20170417T060000
EXDATE;TZID=South Africa Standard Time:20170501T060000

The last method is working fine, but the first method, is causing the toISOstring error.

coderaaron, 2017-04-17:
So this is SUPER hacky but it got my calendar to load.  The only thing is it takes just the first exdate and ignores the rest so consider it an alpha-version, not sure if I'll have to time make it better so if anyone else has the time I'd appreciate the help:

MagicMirror/modules/default/calendar/vendor/ical.js:200

```  // EXDATE is an entry that represents exceptions to a recurrence rule (ex: "repeat every day except on 7/4").
  // There can be more than one of these in a calendar record, so we create an array of them.
  // The index into the array is the ISO string of the date itself, for ease of use.
  // i.e. You can check if ((curr.exdate != undefined) && (curr.exdate[date iso string] != undefined)) to see if a date is an exception.
  var exdateParam = function (name) {
      return function (val, params, curr) {
          var exdate = new Array();
          dateParam(name)(val, params, exdate);
          curr[name] = curr[name] || [];
          date_array = [];
          if(typeof exdate[name]=='string'){
             exdate_array=exdate[name].split(',');
             exdate_string=exdate_array[0];
             exdate[name]=exdate_string.splice(13,0,":").splice(11,0,":").splice(6,0,"-").splice(4,0,"-");
          } else {
             curr[name][exdate[name].toISOString()] = exdate[name];
          }
          return curr;
      }
  }`

DutchDynamics, 2017-04-18:
Hi,

Took your example and played around a bit. This is loading my calendar, and it's including all exceptions not only the first one

`    
// EXDATE is an entry that represents exceptions to a recurrence rule (ex: "repeat every day except on 7/4").
    // There can be more than one of these in a calendar record, so we create an array of them.
    // The index into the array is the ISO string of the date itself, for ease of use.
    // i.e. You can check if ((curr.exdate != undefined) && (curr.exdate[date iso string] != undefined)) to see if a date is an exception.
    var exdateParam = function (name) {
        return function (val, params, curr) {
            var index;
            var tmpDates = new Array();
            var exdate = new Array();

            if (val.indexOf(",") > -1) {
                tmpDates = val.split(",");

                for (index = 0; index < tmpDates.length; ++index) {
                    exdate = new Array();
                    dateParam(name)(tmpDates[index], params, exdate);
                    curr[name] = curr[name] || [];
                    curr[name][exdate[name].toISOString()] = exdate[name];
                }                   
            }
            else {
                dateParam(name)(val, params, exdate);
                curr[name] = curr[name] || [];
                curr[name][exdate[name].toISOString()] = exdate[name];
            }
            return curr;
        }
    }
`

MrXfree11, 2017-04-20:
Hi, any chance to give me an hint which file to edit, to apply that super hacky workarround? The publicly published .ical file from my Exchange 2016 Calendar creates exactly the same error since the MM upgrade:

`Create new news fetcher for url: http://www.faz.net/rss/aktuell/ - Interval: 300000
Whoops! There was an uncaught exception...
TypeError: curr.exdates[i].toISOString is not a function
`

coderaaron, 2017-04-20:
The file to edit is `MagicMirror/modules/default/calendar/vendor/ical.js`

MrXfree11, 2017-04-20:
AWESOME! works like a charm. Thanks a lot. 

Cheers.

mitch1138, 2017-04-22:
_tgeimer_ suggested editing
`ORDINAL_BASE: new Date(1970, 0, 1),`
in
`/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js`
to an earlier date.  I used 1000 and it resolved my issue.

roramirez, 2017-04-22:
I think to be this we need fork the actual repository and apply the patch or create a rules for add this path. cc @fewieden @amcolash

amcolash, 2017-04-23:
I made an issue on the rrule repo [here](https://github.com/arolson101/rrule-alt/issues/4), maybe we can get official change instead of a local version. I was looking into the code and it doesn't look like it would be easy to inject our own `ORDINAL_BASE`. Seems like a patch, fork or local copy are the options.

MichMich, 2017-04-24:
I think the best approach is indeed forking one of the libs and maintaining it ourselves. Any volunteer? 😄

Dup0n7, 2017-04-25:
Where in MagicMirror/modules/default/calendar/vendor/ical.js do you add those lines @DutchDynamics  ?

DutchDynamics, 2017-04-25:
Hi,

I attached the versions of ical.js and node-ical.js that i'm currently using
The change I made is in the method 'exdateParam' in the ical.js
From line 182 till 202

[ical.js.zip](https://github.com/MichMich/MagicMirror/files/955847/ical.js.zip)

roramirez, 2017-04-26:
Note: I've testing the patch proposed before 
https://github.com/roramirez/rrule-alt/commit/7e1cd5892384c5f92c5b9294d72353322767ff72
and  the tests of rrule-alt  are broke inside the lib.

GobleSt, 2017-05-01:
Edits by coderaaron and DutchDynamics fixed this for me also as I had the same exact issue.

Thanks all!!!

deadherring, 2017-05-04:
Thanks @coderaaron and @DutchDynamics!!  I've been having the same problem with the exchange calendar and copied your files to my magic mirror and now it's working!

elmerito25, 2017-05-05:
Same here!  Thanks a lot!

MichMich, 2017-05-26:
@roramirez Can I close this issue?

RamblingGeekUK, 2017-06-19:
I was trying to use ical from outlook.com and it didn't work until I replaced the .js files provided by @coderaaron and @DutchDynamics, thanks!

onedumbtrucker, 2017-08-25:
Fixed it for me :)

MichMich, 2017-09-06:
Closed due to inactivity. Feel free to reopen.

closed 2017-09-06 by MichMich

Munich81, 2018-04-10:
Just ran into the same problem with the current verion. Any reason why the fix didn't make it into master?

lawrence-jeff, 2018-09-08:
Also just ran into this with an outlook.com calendar