MagicMirrorOrg/MagicMirror #841: Microphone is not working

MaengMarco, 2017-04-09:
I did it(npm install && npm start)
But microphone was not working.
on the other hand, If I access '192.168.0.11:8080' on another computer, microphone is working.
What should I do?

MichMich, 2017-04-11:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-04-11 by MichMich

vandenberg1985, 2018-07-27:
I have the same problem.
you can also tray from the LXTerminal
cd Magicmirror
node serveronly (start the server on the background)
*open new LXterminal*
chromium-browser --kiosk http://localhost:8080 (open chrome in fulscreen)
and the microphone works.