MagicMirrorOrg/MagicMirror #847: Meeting subjects don't fully display if they contain apostrophes

rpinyxxjoe, 2017-04-13:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

fewieden, 2017-04-22:
Can you give more details about your issue please, as you didn't fill out any required information.

rpinyxxjoe, 2017-04-22:
Platform: Place your platform here... give us your web browser/Electron version and your hardware Raspberry Pi 3

Node Version: version 0.12.13

MagicMirror Version: newer JavaScript version (v2)

Description: when connecting to Google Calendar's API, user-generated meetings/appointments stop parsing once an apostrophe is detected in the meeting subject. However, system-generated meetings/appointments parse the entire string. Example: when I create a meeting "Mother's birthday", it shows "Mother". For the upcoming holiday, "Mother's Day", the entire string is generated.

Steps to Reproduce: create a meeting in Google Calendar with a meeting subject containing an apostrophe in the string, connect via API and confirm if the Calendar module displays the entire string.

Expected Results: Expect to see the meeting request as shown in Google Calendar.

Actual Results: Meeting request subject string displayed up until the apostrophe character in the string.

Configuration: 

	calendars: [
		{
			url: 'http://link to myGoogleCalendar.ics',
			symbol: 'calendar',
			auth: {
			    user: 'x',
			    pass: 'x',
			    method: 'basic'
			}
		},
	],

Arkkimaagi, 2017-05-17:
It seems that the default calendar module removes certain strings:
```javascript
		titleReplace: {
			"De verjaardag van ": "",
			"'s birthday": ""
		},
```
I do not know why it does this by default but if you specifically set your config to have `titleReplace: {},` like this:
```javascript
...
			module: "calendar",
			position: "top_left",
			config: {
				titleReplace: {},
...
```
After that it should not remove the strings.

I'd probably change the default behaviour to *not* include these string stripping features. What do the maintainers think?

MichMich, 2017-05-26:
I think I agree. Feel free to send a PR.

MichMich, 2017-09-29:
Closed due to inactivity.

closed 2017-09-29 by MichMich