MagicMirrorOrg/MagicMirror #857: 'node ./download-chromedriver.js' fails while installing MagicMirror²

Projekt95, 2017-04-18:
[magic_mirror_debug.txt](https://github.com/electron/chromedriver/files/923713/magic_mirror_debug.txt)

I don't know why this happens.
NPM version: 4.5.0
Node Version: 6.10.2
Raspbian with Pixel + newest Firmware & Updates

Last lines after install fails, full debug.txt is attached:

```
electron-chromedriver@1.6.0 install /root/MagicMirror/node_modules/electron-chromedriver
node ./download-chromedriver.js

module.js:471
    throw err;
    ^

Error: Cannot find module '/root/MagicMirror/node_modules/electron-chromedriver/download-chromedriver.js'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:393:7)
    at startup (bootstrap_node.js:150:9)
    at bootstrap_node.js:508:3
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@1.6.0 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1

```

I tried to install it manually but that didn't work as intended: 
I could avoid the errormessage after I manually cloned the chromedriver repo into node_modules and tried to install it again (`npm install`). But then there's another errormessage that said that the install can't be made with a git module. 
So I removed the git stuff from the new directory and tried it again. This time it worked but then I got a new message: `Failed at the magicmirror@2.1.1 start script ´sh run-start.sh´.`

ExxonLIVE, 2017-04-19:
Hi, i have exactly the same problem.

ExxonLIVE, 2017-04-20:
I think that you can try the server-only version and use your own browser. I'll try that right now.

Projekt95, 2017-04-20:
@ExxonLIVE Did it work for you?

ExxonLIVE, 2017-04-21:
No, it did not. I still had to use "npm install", thats where the error is triggered.

ExxonLIVE, 2017-04-22:
It is working. 

Have you been running as root all the time? I tried it with the "normal" "pi"-user and it worked. (I did it on a  not clean system. I tried some different stuff before. I'll try this on a clean system now.)

ExxonLIVE, 2017-04-22:
It IS working. 

Clean Raspberry Pi IMG, downloaded a week ago. Just run the auto install as PI-user, no update/upgrade. It worked.

Projekt95, 2017-04-22:
@ExxonLIVE 
Nice! Yes I was running the install with sudo.
I'll try your solution ASAP! :)

Projekt95, 2017-04-22:
Without sudo it works like a charm! 
I don't even know how I thought that I have to run the Installer with sudo permissions... lol
Well, thanks. You really helped me out!

closed 2017-04-22 by Projekt95

andreicioromila, 2017-05-14:
Isn't it weird that the MagicMirror install script changes user to root if this is causing problems?

QNimbus, 2017-07-13:
I've had this issue too on Ubuntu 16.04 LTS and in my case I was able to solve the issue by upgrading the default NodeJS installation (version 4.x) to the latest NodeJS v8.x. Hope this can help people finding this through Google search.

roramirez, 2017-07-15:
Yes, NodeJs 4.x is too old and MagicMirror dont have support to work with older version that 5.1.x.

You can find this in https://github.com/MichMich/MagicMirror/blob/master/installers/raspberry.sh#L22 ;)

lordvalium, 2018-01-02:
When i try with sudo i get 
`Error: EACCES: permission denied, mkdir '/home/pi/MagicMirror/node_modules/spectron/node_modules/electron-chromedriver/.electron'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@1.7.1 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron-chromedriver@1.7.1 install script.
`

Idea?

woodjamie, 2018-01-07:
I too am experiencing the same issue whilst trying to upgrade to the latest version of MM. Have tried upgrading node.js from version 6->8 but still did not work.

lordvalium, 2018-01-08:
I did a complete reinstall of MM.

beitaz, 2018-12-01:
When i delete global electron installed by npm, it's gone.

Jiiiiiin, 2019-01-11:
that's how i solved it:

```bash
jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ●  npm i

> fsevents@1.2.4 install /Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app/node_modules/fsevents
> node install

[fsevents] Success: "/Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app/node_modules/fsevents/lib/binding/Release/node-v64-darwin-x64/fse.node" already installed
Pass --update-binary to reinstall or --build-from-source to recompile

> electron-chromedriver@3.0.0 install /Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app/node_modules/electron-chromedriver
> node ./download-chromedriver.js

/Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app/node_modules/electron-chromedriver/download-chromedriver.js:19
  if (err != null) throw err
                   ^

Error: EACCES: permission denied, open '/Users/jiiiiiin/Library/Caches/electron/tmp-2708-1-chromedriver-v3.0.0-darwin-x64.zip'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@3.0.0 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron-chromedriver@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/jiiiiiin/.npm/_logs/2019-01-11T01_57_20_866Z-debug.log
 ✘ jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ●  ll /Users/jiiiiiin/Library/Caches/electron
total 206976
-rw-r--r--  1 root  staff   4.6K  1 10 10:11 SHASUMS256.txt-4.0.1
-rw-r--r--  1 root  staff    48M  1 10 11:42 electron-v3.0.14-darwin-x64.zip
-rw-r--r--  1 root  staff    52M  1 10 10:11 electron-v4.0.1-darwin-x64.zip
 jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ●  rm -rf /Users/jiiiiiin/Library/Caches/electron
rm: /Users/jiiiiiin/Library/Caches/electron/electron-v3.0.14-darwin-x64.zip: Permission denied
rm: /Users/jiiiiiin/Library/Caches/electron/electron-v4.0.1-darwin-x64.zip: Permission denied
rm: /Users/jiiiiiin/Library/Caches/electron/SHASUMS256.txt-4.0.1: Permission denied
rm: /Users/jiiiiiin/Library/Caches/electron: Directory not empty
 ✘ jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ●  sudo rm -rf /Users/jiiiiiin/Library/Caches/electron
Password:
 jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ●  npm i

> electron-chromedriver@3.0.0 install /Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app/node_modules/electron-chromedriver
> node ./download-chromedriver.js

> ynrcc-ebank-desktop-app@0.1.0 postinstall /Users/jiiiiiin/Documents/GitHub/ynrcc-ebank-desktop-app
> electron-builder install-app-deps

  • electron-builder version=20.38.4
  • no native production dependencies
npm WARN ajv-keywords@2.1.1 requires a peer of ajv@^5.0.0 but none is installed. You must install peer dependencies yourself.

added 4 packages from 3 contributors in 13.845s

   ╭───────────────────────────────────────────────────────────────╮
   │                                                               │
   │       New minor version of npm available! 6.4.1 → 6.5.0       │
   │   Changelog: https://github.com/npm/cli/releases/tag/v6.5.0   │
   │               Run npm install -g npm to update!               │
   │                                                               │
   ╰───────────────────────────────────────────────────────────────╯

 jiiiiiin@jiiiiiins-MacBook-Pro  ~/Documents/GitHub/ynrcc-ebank-desktop-app   develop ● 
```